mne-tools/mne-python #3134: Importing OpenBCI data

ncullen93, 2016-04-16:
Any plan to read/stream OpenBCI data? I use that hardware... so I'd be interested in incorporating it. Perhaps something along the lines of the following project: https://github.com/OpenBCI/OpenBCI_Python  or ideally just incorporating that project in MNE.

teonbrooks, 2016-04-16:
hi @ncullen93, we have a pull request, https://github.com/mne-tools/mne-python/pull/2671, that aims to incorporate OpenBCI streaming into MNE. feel free to join the discussion there, and if you would like, you can help contribute to help finish. we can re-stimulate its progress. 
there was an older discussion before https://github.com/mne-tools/mne-python/pull/2462 about this. the data file that comes from OpenBCI_Processing GUI is a modified csv. we decided that it would not be advantageous to support it within the package for a host of reasons. 
I wrote a reader for converting the Processing Data for MNE [here](https://github.com/OpenBCI/OpenBCI_Python/blob/master/externals/mne_openbci.py). 

Let me know if you have any more questions about this. I would happy to answer them. If not, feel free to close this issue.

ncullen93, 2016-04-17:
I appreciate the info and apologize for the repeat - should have looked harder. Will close this.

closed 2016-04-17 by ncullen93

teonbrooks, 2016-04-17:
no worries :)

d3christ, 2018-06-08:
Hey when I try to run the "from mne_openbci import read_raw_openbci"
I get an error saying that mne is needed but I already have mne installed and activated.

cbrnr, 2018-06-11:
This module tries to import `_BaseRaw`, which doesn't exist (anymore?) in MNE. Try changing this in lines 17 and 21 to `BaseRaw` and it should work. @teonbrooks you might want to update your module. Also, I don't really understand why you set `np =  None` and `mne = None`?

teonbrooks, 2018-06-12:
@d3christ that mne_openbci package is deprecated. are you trying to read data in real time or are you trying to the export csv file?

teonbrooks, 2018-06-12:
I wrote the package like three years and haven't used it since. both mne and openbci has changed significantly since then