mne-tools/mne-python #3135: Proj example failing on master

Eric89GXL, 2016-04-18:
From #3119:
```
larsoner@bunk:~/custombuilds/mne-python$ python examples/visualization/plot_ssp_projs_sensitivity_map.py 
==================================
Sensitivity map of SSP projections
==================================

This example shows the sources that have a forward field
similar to the first SSP vector correcting for ECG.

Traceback (most recent call last):
  File "examples/visualization/plot_ssp_projs_sensitivity_map.py", line 31, in <module>
    ssp_ecg_map = sensitivity_map(fwd, ch_type='grad', projs=projs, mode='angle')
  File "/home/larsoner/custombuilds/mne-python/mne/proj.py", line 377, in sensitivity_map
    co = linalg.norm(np.dot(gg[:, 2], U))
ValueError: shapes (204,) and (0,) not aligned: 204 (dim 0) != 0 (dim 0)
```
I'll try to take a look to see what's going on.

jaeilepp, 2016-04-18:
I have a fix ready for it. Just a minute.

Eric89GXL, 2016-04-18:
There needs to be a fix at the `sensitivity_map` level and the example level, I'm working on both. I'll stop if you already have both done...

jaeilepp, 2016-04-18:
Okay I'm not sure what's the problem in the sensitivity map. The example fix is just not removing the grad projector.

jaeilepp, 2016-04-18:
``projs = projs[3:][::2]  # take only one projection per channel type`` ->
``projs = projs[::2]  # take only one projection per channel type``

Eric89GXL, 2016-04-18:
cool I'll put that in

PR #3136 (2016-04-18): MRG: Fix proj
Closes #3135.
closed 2016-04-18 by jaeilepp