mne-tools/mne-python #3138: Picking a small subset of channels when creating epochs objects can zero out MAG channels

Teekuningas, 2016-04-18:
Like this:

```python
import mne
import numpy as np

data_path = mne.datasets.sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'

raw = mne.io.Raw(raw_fname, preload=True)

events = mne.find_events(raw)

epochs = mne.Epochs(raw, events, 1, -0.2, 0.5, picks=[2], preload=True)

assert not np.allclose(epochs._data, np.zeros((72, 1, 106)))
```

Is this a feature? It has something to do with projections.

dengemann, 2016-04-18:
Are you sure that the data are not at a scale below the tolerance?
Can you plot the data? Are they actually zero?

Eric89GXL, 2016-04-18:
It does have to do with projections, see #2310. Try it with `proj=False` or better do `raw.add_proj([], remove_existing=True)` and you should be okay.

On latest master you should actually see a warning about projection being dangerous, do you?

Teekuningas, 2016-04-18:
That way I seem to be okay. But what if I wanted that the resulting epochs would be as if I had created the epochs without picks and with projections normally applied, and only afterwards dropped the channels? Is there a better way than this:
```python
proj_raw = raw.copy().apply_proj()
proj_raw.info['projs'] = []
epochs = mne.Epochs(proj_raw, ..., proj=True, preload=True)
del proj_raw
```
Obviously if I was doing just that, I could do as I said and create without picks and drop channels afterwards, but the reason for using picks was that I'd like that reject parameters would only be taken into consideration on those channels listed in picks-list, as other channels do not really matter for the later analysis.. There might be a better way for this too..?

Eric89GXL, 2016-04-18:
Could you let me know if this is true about your original code...

> On latest master you should actually see a warning about projection being dangerous, do you?

In other words, hopefully you had some warning of danger at least.

Currently what you suggest about overwriting `raw.info['projs']` manually is the best way to accomplish what you want.

Teekuningas, 2016-04-18:
> On latest master you should actually see a warning about projection being dangerous, do you?

The software, that I'm working with, has only 0.11.0 as a dependency at the moment, so I didn't get the warning then, but with the newest master it truly does warn me, once for each projection. : ]

I suppose this can be closed now, right?

Eric89GXL, 2016-04-18:
GitHub auto-links it to the issue I mentioned above, so we should be able to keep it in mind going forward with that issue. So sure

closed 2016-04-18 by Teekuningas