mne-tools/mne-python #3142: Error reading BTI file (header, file type empty)

jona-sassenhagen, 2016-04-19:
This is for HCP subject 195041, run 0, story/math. Reading the BTI file fails because the function to read the file_type for the header encounters an empty string.

```
raw_fid, config_fid, convert=convert, head_shape_fname=None,
         sort_by_ch_name=False, rename_channels=False, preload=True)

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in read_raw_bti(pdf_fname, config_fname, head_shape_fname, rotation_x, translation, convert, rename_channels, sort_by_ch_name, ecg_ch, eog_ch, preload, verbose)

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)
    610 
    611 

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in read_raw_bti(pdf_fname, config_fname, head_shape_fname, rotation_x, translation, convert, rename_channels, sort_by_ch_name, ecg_ch, eog_ch, preload, verbose)
   1378                   convert=convert, rename_channels=rename_channels,
   1379                   sort_by_ch_name=sort_by_ch_name, ecg_ch=ecg_ch,
-> 1380                   eog_ch=eog_ch, preload=preload, verbose=verbose)

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in __init__(self, pdf_fname, config_fname, head_shape_fname, rotation_x, translation, convert, rename_channels, sort_by_ch_name, ecg_ch, eog_ch, preload, verbose)

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/utils.py in verbose(function, *args, **kwargs)
    607         with use_log_level(verbose_level):
    608             return function(*args, **kwargs)
--> 609     return function(*args, **kwargs)
    610 
    611 

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in __init__(self, pdf_fname, config_fname, head_shape_fname, rotation_x, translation, convert, rename_channels, sort_by_ch_name, ecg_ch, eog_ch, preload, verbose)
   1026             translation=translation, convert=convert, ecg_ch=ecg_ch,
   1027             rename_channels=rename_channels,
-> 1028             sort_by_ch_name=sort_by_ch_name, eog_ch=eog_ch)
   1029         self.bti_ch_labels = [c['chan_label'] for c in bti_info['chs']]
   1030         # make Raw repr work if we have a BytesIO as input

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in _get_bti_info(pdf_fname, config_fname, head_shape_fname, rotation_x, translation, convert, ecg_ch, eog_ch, rename_channels, sort_by_ch_name)
   1119     logger.info('Reading 4D PDF file %s...' % pdf_fname)
   1120     bti_info = _read_bti_header(
-> 1121         pdf_fname, config_fname, sort_by_ch_name=sort_by_ch_name)
   1122 
   1123     dev_ctf_t = Transform('ctf_meg', 'ctf_head',

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in _read_bti_header(pdf_fname, config_fname, sort_by_ch_name)
    896     """
    897 
--> 898     info = _read_bti_header_pdf(pdf_fname) if pdf_fname is not None else dict()
    899     cfg = _read_config(config_fname)
    900     info['bti_transform'] = cfg['transforms']

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/bti/bti.py in _read_bti_header_pdf(pdf_fname)
    833         # actual header starts here
    834         info = {'version': read_int16(fid),
--> 835                 'file_type': read_str(fid, 5),
    836                 'hdr_size': start - header_position,  # add for convenience
    837                 'start': start}

/home/jona/anaconda3/lib/python3.5/site-packages/mne-0.12.dev0-py3.5.egg/mne/io/utils.py in read_str(fid, count)
    177     dtype = np.dtype('>S%i' % count)
    178     string = fid.read(dtype.itemsize)
--> 179     data = np.fromstring(string, dtype=dtype)[0]
    180     bytestr = b('').join([data[0:data.index(b('\x00')) if
    181                           b('\x00') in data else count]])

IndexError: index 0 is out of bounds for axis 0 with size 0
```

Or, in abbreviated form:

```
from mne.externals.six import b

with mne.io.bti.bti._bti_open(pdf_fname, 'rb') as fid:
    fid.seek(-8, 2)
    fid.seek(header_position, 0)
    count=5
    dtype = np.dtype('>S%i' % count)
    string = fid.read(dtype.itemsize)
    data = np.fromstring(string, dtype=dtype)[0]

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-123-8f2ed34e21d4> in <module>()
      7     dtype = np.dtype('>S%i' % count)
      8     string = fid.read(dtype.itemsize)
----> 9     data = np.fromstring(string, dtype=dtype)[0]
     10     bytestr = b('').join([data[0:data.index(b('\x00')) if
     11                           b('\x00') in data else count]])

IndexError: index 0 is out of bounds for axis 0 with size 0
```

where string = `b''`

@dengemann

dengemann, 2016-04-19:
This may happen if your files did not download correctly. There is no MNE
issue here I bet.

jaeilepp, 2016-04-19:
There is a fix for that in https://github.com/mne-tools/mne-python/pull/3131

jona-sassenhagen, 2016-04-19:
Redownloaded twice and now it works. Thanks!

closed 2016-04-19 by jona-sassenhagen