mne-tools/mne-python #3160: make_fixed_length_events adds first_samp to start  indices

jhouck, 2016-04-23:
In event.py, make_fixed_length_events always adds first_samp to the start index, i.e.:
`start = raw.time_as_index(start)`
`start = start[0] + raw.first_samp`
This isn't obvious unless you look at the function.  It might be problematic if, for instance, the start time is based on event indices from find_events, which will have already been adjusted for first_samp (via _find_events, via _find_stim_steps).  

Is this an actual issue or maybe just worth a mention in the function's help?

Eric89GXL, 2016-04-23:
We could add a `first_samp=True` kwarg, that would make it explicit and easy to avoid. WDYT?

jhouck, 2016-04-23:
Sounds good to me.

Eric89GXL, 2016-04-23:
If you have time to make a PR feel free, otherwise I can find time next week

jhouck, 2016-04-25:
OK, PR created.

closed 2016-04-25 by jhouck