mne-tools/mne-python #3163: 0.12

Eric89GXL, 2016-04-23:
Planning to push out a release of 0.12 next week (April 29) according to our internal deadlines. Here is what is currently marked to go in:

https://github.com/mne-tools/mne-python/milestones/0.12

If anyone wants other things in, feel free to comment here, especially if you are willing to make PRs to make it happen :)

jona-sassenhagen, 2016-04-23:
Please do add #3161.

dengemann, 2016-04-24:
Let's add the GFP/VIZ plot and JR's fix for the cv behavior in TimeGen.

Eric89GXL, 2016-04-29:
@dengemann it looks like there are three outstanding PRs that require your attention before we can release: #3144, #3143, and #3123. Will you still have time today, or should we move the release to next week to give you more time?

dengemann, 2016-04-29:
Ok will look into them now, let's see how far we get.

Eric89GXL, 2016-04-29:
From talking to @dengemann and @agramfort it would be safest to move this to next week actually. (I somehow didn't realize until now that the deadline we had for release coincided with the deadline for Biomag 2016 abstract submissions...). We're going to shoot for this coming Wednesday May 4th for release instead.

jhouck, 2016-04-29:
FWIW, the Biomag submission deadline was moved to May 14.

Eric89GXL, 2016-04-29:
Thanks for the heads up.

Too bad we don't have an acausal filter we can apply to @dengemann to let past him know last week :)

Eric89GXL, 2016-05-04:
We're going to delay this another couple of days so @jaeilepp can help us with doc updates.

jaeilepp, 2016-05-05:
You mean the stable docs? I can take care of it today.

agramfort, 2016-05-05:
looks like we have a couple more bugs to fix before the release.

thanks

closed 2016-05-27 by Eric89GXL