mne-tools/mne-python #3164: Is there any pattern behind when times are in ms or in s?

jona-sassenhagen, 2016-04-23:
Try this:

```
%matplotlib inline

import matplotlib.pyplot as plt

import mne
from mne.datasets import sample
from mne.stats.regression import linear_regression_raw

# Load and preprocess data
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, preload=True).pick_types(
    meg='grad', stim=True, eeg=False).filter(1, None, method='iir')

# Set up events
events = mne.find_events(raw)
event_id = {'Aud/L': 1, 'Aud/R': 2}
tmin, tmax = -.1, .5

# regular epoching
picks = mne.pick_types(raw.info, meg=True)
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, reject=None,
                    baseline=None, preload=True, verbose=False)

epochs["Aud"].average().plot_joint(times=[.1, .15], ts_args={'xlim':(-.1, .4)});
```

Or rather, DON'T do this, because the last step sometimes just results in a very silly plot, but sometimes takes a very long time to run because it is trying to plot something very weird.

Instead, you must plot as:

```
epochs["Aud"].average().plot_joint(times=[.1, .15], ts_args={'xlim':(-100, 400)});
```

This is arguably very counterintuitive.

Is there any chance we can get this sorted out at all, or are we locked into this heterogeneity by legacy?

jona-sassenhagen, 2016-04-23:
To clarify, this is not about `plot_joint`, which simply passes parameters to evoked.plot and evoked.plot_topomap.

Eric89GXL, 2016-04-24:
I think in all newer code we try to use `s` and `m` but in some older legacy places we have `ms` and/or `mm`. IMO it would be worth going through a deprecation cycle with `time_units='ms' -> 's'`, it's what we're doing with `fit_sphere_to_headshape` [here] (https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L826).

agramfort, 2016-04-24:
well I think we can live with : 'xlim':(-100, 400)
it does not say time and the data is plotted in x axis in milliseconds
that's why

jona-sassenhagen, 2016-04-24:
>well I think we can live with : 'xlim':(-100, 400)
>it does not say time and the data is plotted in x axis in milliseconds
>that's why

Definitely not least-surprise though you have to admit.

>IMO it would be worth going through a deprecation cycle with time_units='ms' -> 's'

Yes! I can go through the code and see how big of a deal it would be.

agramfort, 2016-04-24:
I don't really like the idea of having complex API handling time units...

personally I think we can live with the current API

jona-sassenhagen, 2016-05-11:
I just had my browser crash on me because I set xlim in seconds. (In the Notebook, it creates a massive PNG in that case.)

I will at least try to think of avoiding this situation, e.g. by fixing plot_joint for situations where times is not in xlim or xlim is very short.

Eric89GXL, 2016-05-11:
Closing as "won't fix" until we convince @agramfort it's worthwhile :)

But yes if you can think of a good way to warn people at least that would help.

larsoner, 2018-04-02:
This continues to come up from time to time. And from working on XXX it looks like the only plots that use `[ms]` are `evoked.plot`. This is inconsistent even with things that produce `evoked.plot`-like sensor trace plots. :(

@agramfort any change of heart about adding `time_unit` to `evoked.plot` and `evoked.plot_joint`? I think it would be a couple dozen lines of code to do it.

I'd actually be in favor of a deprecation cycle to get to `'s'`. It would be painful but then everything would be unified.

larsoner, 2018-04-02:
And by "working on XXX" I meant "working on #5079" ... that's what I get for drafting a reply before actually making the PR :)

agramfort, 2018-04-02:
so "Time (s)" everywhere?

larsoner, 2018-04-02:
Yes, we are 98% there, just `evoked.plot` to conquer IIRC

PR #5082 (2018-04-08): MRG: Unify around time_unit="s"
Closes #3164.

This moves `evoked.plot` and `evoked.plot_topomap` toward using `time_unit='s'` as the default, consistent with all other MNE plotters. This means adding the param to both, deprecating `ms`->`s`, and also deprecating the `scaling_time` parameter in `evoked.plot_topomap`.

This will emit some warnings in examples and tests, I hope to fix most of them after inspecting CI output.

@jona-sassenhagen @mmagnuski @cbrnr agree this is a good plan?
closed 2018-04-08 by agramfort