mne-tools/mne-python #3168: ICA tutorial: wrong components (random seed?)

jona-sassenhagen, 2016-04-24:
The ICA tutorial is a *bit* off.

[image]

[image]

I can't promise I can fix this for 0.12 ...

@dengemann

dengemann, 2016-04-24:
I will. No worries.

jaeilepp, 2016-04-26:
Fixed in https://github.com/mne-tools/mne-python/pull/3174. See if you agree.

jona-sassenhagen, 2016-05-11:
Oh oh ...

[image]

I could take care of it eventually.

jona-sassenhagen, 2016-05-11:
@Eric89GXL reopen please :(

Eric89GXL, 2016-05-11:
This was recently fixed. Do you know the cause? Was there a regression,
errant commit, or is this example fragile? (Will reopen when I'm off mobile)

dengemann, 2016-05-11:
This is very weird indeed.

Eric89GXL, 2016-05-11:
Bisecting now...

dengemann, 2016-05-11:
the fun thing is that this got fixed 2 times at least.

Eric89GXL, 2016-05-11:
Found it -- @dengemann this commit makes things non-deterministic, which is funny since it's title is "fix seed issue in ICA tutorial":

https://github.com/mne-tools/mne-python/commit/4205bf09fec99067d4c5da667242146e01a10c94#diff-862a5a4a87f90baa4d64932537a53851R154

Notably, the `seed=` kwarg was removed in that commit :(

Eric89GXL, 2016-05-11:
... or rather, the "reference ICA" has no seed, so when (in that commit) it was switched from using the first of the section-based ICAs (which were deterministic) to the "reference ICA", it lost its deterministic-ness.

Eric89GXL, 2016-05-11:
In any case, I'm not sure of the proper fix here (seed the original, or go back to seeding the others and use the first one again), but I'm not surprised it's non-deterministic, because the random state isn't ever set. @dengemann can you fix it?

dengemann, 2016-05-11:
Ok I'll fix it once more. ... I hope for the last time.

Eric89GXL, 2016-05-13:
@dengemann will you have time for this soon, or should someone else have a look (@jona-sassenhagen maybe?)

dengemann, 2016-05-14:
I will have time on Monday for some MNE hacking.

Eric89GXL, 2016-05-23:
@dengemann it's been a while, any time for this one or should we get someone else on it?

closed 2016-05-27 by agramfort