mne-tools/mne-python #3175: raw.add_channels crashes with ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

dmalt, 2016-04-26:
Hi!

I'm trying to merge some channels from my MEG recordings:
`eog_channels = raw.pick_channels(ch_names=['EMG065'], copy=True)
ecg_channels = raw.pick_channels(ch_names=['ECG063'], copy=True)
merged_channels = ecg_channels.add_channels([eog_channels], copy=True)`

But it add_channels crashes with ValueError:
`---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-28-c0ce0f336ed8> in <module>()
      1 eog_channels = raw.pick_channels(ch_names=['EMG065'], copy=True)
      2 ecg_channels = raw.pick_channels(ch_names=['ECG063'], copy=True)
----> 3 merged_channels = ecg_channels.add_channels([eog_channels], copy=True)

/home/meg/anaconda2/lib/python2.7/site-packages/mne/channels/channels.pyc in add_channels(self, add_list, copy)
    542         data = np.concatenate(data, axis=con_axis)
    543         infos = [self.info] + [inst.info for inst in add_list]
--> 544         new_info = _merge_info(infos)
    545 
    546         # Now update the attributes

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in _merge_info(infos, verbose)

/home/meg/anaconda2/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in _merge_info(infos, verbose)
   1295 
   1296     for k in other_fields:
-> 1297         info[k] = _merge_dict_values(infos, k)
   1298     info._check_consistency()
   1299     return info

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in _merge_dict_values(dicts, key, verbose)

/home/meg/anaconda2/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in _merge_dict_values(dicts, key, verbose)
   1185     # dict
   1186     elif _check_isinstance(values, dict, all):
-> 1187         is_qual = _is_equal_dict(values)
   1188         if is_qual:
   1189             return values[0]

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in _is_equal_dict(dicts)
   1146         # import pdb; pdb.set_trace()
   1147         is_equal.append(all(np.all(k == k0) and
-> 1148                         np.all(v == v0) for k, v in d))
   1149     return all(is_equal)
   1150 

/home/meg/anaconda2/lib/python2.7/site-packages/mne/io/meas_info.py in <genexpr>((k, v))
   1146         # import pdb; pdb.set_trace()
   1147         is_equal.append(all(np.all(k == k0) and
-> 1148                         np.all(v == v0) for k, v in d))
   1149     return all(is_equal)
   1150 

ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()`

Is there something I'm missing? What should I do to append channels to one another?

agramfort, 2016-04-26:
can you replicate this with sample?

if so can you share a full gist to debug? on gist.github.com

closed 2016-04-28 by agramfort

dmalt, 2016-04-28:
Thanks for such a quick fix!
Now it works fine.