mne-tools/mne-python #3179: mne.viz.plot_source_estimates generates blank window with error message

gladiach, 2016-04-26:
I ran mne.viz.plot_source_estimates(stc,subject='<subject_name>',...) but I got a blank window and this error. I get a similar error when I try stc.plot(). This is the error:

(<unknown>:39799): Gtk-WARNING **: gtk_disable_setlocale() must be called before gtk_init()
Traceback (most recent call last):
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 3035, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-1-16aa827441a7>", line 1, in <module>
    mne.viz.plot_source_estimates(stc,subject='eeganes03',surface='inflated',hemi='lh',colormap='auto',time_label='time=%0.2fms',smoothing_steps=10,transparent=None,alpha=1.0,time_viewer=False,config_opts=None,subjects_dir=None,figure=None,views='lat',colorbar=True,clim='auto')
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/mne/viz/_3d.py", line 593, in plot_source_estimates
    brain = Brain(subject, hemi, surface, **kwargs)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/surfer/viz.py", line 395, in __init__
    self._scene_size, offscreen)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/surfer/viz.py", line 212, in _make_viewer
    figure = mlab.figure(title, size=(w, h))
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/mayavi/tools/figure.py", line 86, in figure
    engine.new_scene(name=name, size=size)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/apptools/scripting/recordable.py", line 45, in _wrapper
    result = func(*args, **kw)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/mayavi/core/engine.py", line 434, in new_scene
    viewer = self.scene_factory(**factory_kwargs)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/mayavi/core/ui/mayavi_scene.py", line 89, in viewer_factory
    viewer.open()
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/pyface/i_window.py", line 145, in open
    self._create()
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/pyface/ui/wx/application_window.py", line 151, in _create
    contents = self._create_contents(body)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/tools/ivtk.py", line 392, in _create_contents
    self.scene = self._scene_factory(parent)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/mayavi/core/ui/mayavi_scene.py", line 68, in mayavi_scene_factory
    s = MayaviScene(parent, stereo=p['stereo'])
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/ui/wx/decorated_scene.py", line 69, in __init__
    super(DecoratedScene, self).__init__(parent, **traits)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/ui/wx/scene.py", line 302, in __init__
    super(Scene, self).__init__(parent, **traits)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/tvtk_scene.py", line 201, in __init__
    self.control = self._create_control(parent)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/ui/wx/decorated_scene.py", line 103, in _create_control
    self._panel)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/ui/wx/scene.py", line 575, in _create_control
    stereo=self.stereo)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/tvtk/pyface/ui/wx/wxVTKRenderWindowInteractor.py", line 186, in __init__
    attribList=attribList)
  File "/usr/pubsw/packages/python/anaconda/lib/python2.7/site-packages/wx-3.0-gtk2/wx/glcanvas.py", line 106, in __init__
    _glcanvas.GLCanvas_swiginit(self,_glcanvas.new_GLCanvas(*args, **kwargs))
TypeError: Argument given by name ('attribList') and position (3)

dgwakeman, 2016-04-26:
Hi,

We are indeed having issues with the Martinos Center central python install. Both @haribharadwaj and I have tried to solve it. I posted to both the conda github and conda google groups, but haven't heard anything back:
https://github.com/conda/conda/issues/2322

Thoughts?

d

Eric89GXL, 2016-04-26:
I find that in some corner cases Anaconda can get in a bit of a fragile state. I would try:

1. Downloading the latest Anaconda installer
2. Installing it to some sandbox location like ~/temp
3. `conda update conda`
4. Installing all packages you need (from the `conda` side anyway)

If that works, then I'd go back to your real Anaconda installation, and uninstall all conflicting packages you have listed, then request to install all packages. If that doesn't work, you can figure out why by the difference in package lists between the two setups.

If it doesn't work, [look at what we do for the 2.7-full build on Travis] (https://github.com/mne-tools/mne-python/blob/master/.travis.yml) and follow suit. That one gets installed about 20 times per day and works -- but every couple of months it seems we have to make some small tweak in version numbers to work around Anaconda's imperfections. Travis runs Ubuntu and you're probably CentOS (?) so that might matter, but hopefully not.

jasmainak, 2016-04-26:
It sounds like a mayavi install issue to me. Make sure you use `conda install mayavi`. Also, you don't need sudo access to install anaconda.

gladiach, 2016-04-26:
My colleague helped me fix my issue by adding a file called .tcshrc to my home folder. The file includes the lines:

setenv QT_API pyqt
setenv ETS_TOOLKIT qt4

Now mne.viz works for me.

agramfort, 2016-04-27:
do we have this somewhere in the doc?

if not someone please add and then close this issue.

thanks

jaeilepp, 2016-04-27:
Would this be material for FAQ? Or should we even have a new 'troubleshooting' section. Do we have enough material for that?

agramfort, 2016-04-27:
I would put this in how to setup your python environment in getting started

closed 2016-04-28 by Eric89GXL