mne-tools/mne-python #3180: plot_topomap times for CSP example

teonbrooks, 2016-04-26:
with the plot_decoding_sp_space example, the plot_topomap goes behind the time range of the data. seems like a bug with how times is defined in the example.

http://martinos.org/mne/stable/auto_examples/decoding/plot_decoding_csp_space.html

jaeilepp, 2016-04-27:
I was going to fix this in https://github.com/mne-tools/mne-python/pull/3174 using the new possibility to plot single data points, but then I noticed that ``plot_topomap`` accepts ``axis`` instead of ``axes`` like the rest of the plotting functions. I'm wondering if I can change this to ``axes`` without deprecation cycle.

agramfort, 2016-04-27:
I fear you need to deprecate.

nice catch

Eric89GXL, 2016-04-27:
It shouldn't be too bad a deprecation though:

1. put `axes` in the same point in the arg list as `axis`
2. move `axis=None` to the end
3. if `axis is not None` warn and set `axes=axis`
4. if `axes` is an instance of Axes, make it `[axes]`

right?

teonbrooks, 2016-04-27:
can't take all the credit. thanks @lauragwilliams for pointing this out :)

closed 2016-04-28 by Eric89GXL