mne-tools/mne-python #3183: set_bipolar_reference, keep_old param?

jona-sassenhagen, 2016-04-27:
How about adding a keep_old parameter to `set_bipolar_reference`?

For example, I just created a bipolar montage for a virtual VEOG channel for the purpose of ICA component labelling. However, I did not want to remove the original channels afterwards, for example, because the ICA contained, and should contain, the original channels, not the VEOG. This was approximately 10 lines of fairly messy code all in all that partially replicated regular MNE code that could be greatly simplified if there was an option in `set_bipolar_reference` to keep the old anode and cathode. 

Code wise, I guess it's not too complicated - something like
```
    # Perform bipolar referencing
    for an, ca, name, info in zip(anode, cathode, ch_name, new_ch_info):
        add_reference_channels(inst, name)
        _, bipolar_data = _apply_reference(inst, [ca], [an])
        if isinstance(inst, Evoked):
           inst.data[-1, :] += bipolar_data
        if isinstance(inst, _BaseRaw):
            inst._data[-1, :] += bipolar_data
        elif isinstance(inst, _BaseEpochs):
           inst._data[:, -1, :] += bipolar_data
        info['ch_name'] = name
        inst.info['chs'][-1] = info
        logger.info('Bipolar channel added as "%s".' % name)
    inst.info._update_redundant()

    if not keep_old:
        inst.drop_channels(cathode + anode)
```
That's only like 4 lines more than the current code.

jona-sassenhagen, 2016-04-27:
Maybe it'd be sufficient to use raw.add_channels for this ...

Eric89GXL, 2016-04-27:
How hard would it be to use `raw.add_channels`? If it's a few lines fine, but if bipolar referencing is common we should try to support it without requiring too much code from the user.

jona-sassenhagen, 2016-04-27:
Beat you!

I will have to check.

jona-sassenhagen, 2016-04-27:
Ah I remember. I tried that yesterday.

```
        eog = bipolar(
                raw.copy(), ch_name=ch, copy=False,
                anode="LO1", cathode="SO2").drop_channels(
            [x for x in raw.ch_names if x != ch])
        raw.add_channels([eog_raw])
```

gives

```
RuntimeError: Don't know how to merge 'custom_ref_applied'. Make sure values are compatible.
```

And okay, you can work around that ... but that's where the lines come from.