mne-tools/mne-python #3184: ENH: Reconstruction?

Eric89GXL, 2016-04-27:
Would people be interested in looking at the extent to which e.g. Evoked data can be reconstructed following projections (un-doing the induced signal bias)? Here is the idea:
```
import os.path as op

import mne
from mnefun import plot_reconstruction

data_path = mne.datasets.sample.data_path()
raw = mne.io.read_raw_fif(op.join(data_path, 'MEG', 'sample',
                          'sample_audvis_raw.fif'), preload=True)
raw.pick_types(meg=True, eeg=False, eog=True, stim=True, exclude=())
events = mne.find_events(raw, 'STI 014')
raw.add_proj(mne.preprocessing.compute_proj_eog(raw)[0])
raw.add_proj(mne.preprocessing.compute_proj_ecg(raw)[0])
epochs = mne.Epochs(raw, events, {'Left auditory': 1}, -0.2, 0.5,
                    baseline=(None, 0), proj='delayed',
                    reject=dict(grad=4000e-13, mag=4e-12, eog=150e-6))
evoked = epochs.average()
evoked.plot_reconstruction()
```
Would give you something like:

[image]

The "reconstructed" data is obtained by projecting the (SSP-projected) data to a sphere using MNE and coming back out. The critical step is that the projection operator is applied to the lead fields as well as the data, thereby undoing the bias to the data.

In Xfit, this is known as reconstruction "on" vs "off", and helps visualize whether or not your projectors have eliminated a true signal of interest. Techniques like dipole fitting and minimum-norm estimates both do this type of compensation during estimation, so it gives you a better idea in sensor space of how much of your signal of interest is still there.

agramfort, 2016-04-27:
nice.

this should be made public. We'll see what's the best way to expose this.

Eric89GXL, 2016-04-27:
Cool, will do.

It's pretty straightforward from an implementation standpoint, *except* that right now we only have `_map_meg_channels` using MNE -- `_map_eeg_channels` uses spherical splines which will not work (not sure why we didn't use MNE for that?). So i'll need to add a `method` arg to `_map_eeg_channels`, then it should only be ~30 lines or so for the viz function.

PR #8033 (2020-07-22): MRG, ENH: Add proj="reconstruct" option
Closes #3184
closed 2020-07-22 by larsoner