mne-tools/mne-python #3186: mne.read_events not reading all events

drammock, 2016-04-27:
On current master, `mne.read_events` is omitting 20 rows.  MWE is below.  The file "sample-eve.txt" is [here](http://dan.mccloy.info/data/sample-eve.txt) and was generated by `mne.write_events`.

```python
import mne
import pandas

mne_events = mne.read_events('sample-eve.txt')
pandas_df = pandas.read_csv('sample-eve.txt', sep=r'\s*', header=None)

print(mne_events.shape)  # (4620, 3) -- 20 rows missing
print(pandas_df.shape)   # (4640, 3)
```

drammock, 2016-04-27:
oh, I get it.  I have re-coded the event values during processing to replace all `1` events with a stimulus code reconstructed from a string of preceding `4` and `8` triggers.  20 of those recoded event values are `0`, and default is `mask=0` so they're getting (silently) skipped.  Unfortunately `mask=-1` doesn't work, nor does `mask=None`.  Is my only option to re-do my stimulus codes?  That is very inconvenient, as they act as indices into several data structures.

Eric89GXL, 2016-04-27:
One problem with a code of `0` is that, in the first line, it has a special meaning (or used to?) for the C tools, so it goes beyond the `mask` problem, see:

https://github.com/mne-tools/mne-python/blob/master/mne/event.py#L260

I suppose we could add a `mask=None` argument for this use case, and state clearly in the docstring that it could cause problems if events are used by the C tools.

dgwakeman, 2016-04-28:
@drammock, I'm not sure I see the problem, with when you do the re-coding to set those instead of to 0 to 1023 or some other value?

Eric89GXL, 2016-04-28:
(Personally I do a uniform +1 shift and later -1 shift as needed, but it is annoying in some cases)

drammock, 2016-04-28:
@dgwakeman of course, that is a work-around I considered (as is Eric's suggestion to shift all events by 1).  But such workarounds strike me as inelegant and error-prone. As someone who has only rarely used the C API and will likely never do so again, the [zero-as-a-special-case](https://github.com/mne-tools/mne-python/blob/master/mne/event.py#L260) that @Eric89GXL pointed out strikes me as, if not cruft, then at least something that should arguably not be the default, and at a bare minimum should at least be preventable via `mask=None` or some equivalent means.

drammock, 2016-04-28:
(if devs/BDFL won't agree even to that, then perhaps at least a `warning` or `info` when reading in the events, saying how many were read and how many were skipped?)

dgwakeman, 2016-04-28:
I guess, I should of asked how the recoding is being done (personally, I've never needed to recode, I use the mask option and choose the values I want when I design the experiment). Maybe, we need better code for recoding stimuli? It seems pretty silly to have an event entry that reads `time 0 0`, if we stop creating events that are `time 0 0` it should avoid the problem. How will some other researcher read an event where the trigger went from 0 -> 0

Eric89GXL, 2016-04-28:
> I should of asked how the recoding is being done

There are many potential use cases for re-coding event numbers -- just think about e.g. changing them based on subject response correctness, correcting values that were incorrect during acquisition, or inserting sub-trial-level stamps that weren't put in during the trial (e.g., word onsets in a continuous sentence). So I'm not really sure the particular re-coding motivation is relevant, since we don't want to force people to only use the triggers that the MEG machine recorded, plus masks later. But if you're just curious, we use a serially-stamped (before trial start) ID to avoid some hardware issues we had a long time ago at MGH, and are being paranoid that they might reoccur, so all our trials actually start with a 1 trigger, and thus need to be re-coded to be meaningful.

> How will some other researcher read an event where the trigger went from 0 -> 0

In my mind, event values are no longer this physical change you're talking about related to changes on TTL lines, but rather just numbers that are used to control the abstractions about an experiment. So 1 and 2 and 3 and even 0 could have meaning once you're at the point of reading and writing event lists. Somewhere in the code, 1, 2, 3 and just as easily (well maybe not currently) 0 are given more meaning by the `event_id` specific to the paradigm.

> then perhaps at least a warning or info when reading in the events

I agree at a minimum we need to warn when there are `zero` events. At this point it is probably reasonable to do it even for the zeroes that appear in the first line -- people working with legacy event lists will get a warning, but will hopefully know to disregard it.

agramfort, 2016-04-28:
+1 for warning

PR #3197 (2016-05-07): MRG: warn when read_events drops/masks events; allow mask=None
Closes #3186
closed 2016-05-07 by agramfort