mne-tools/mne-python #3187: BUG: cwt_morlet decim

kingjr, 2016-04-28:
```
import numpy as np
from mne.time_frequency import cwt_morlet
X = np.random.rand(10, 200)
tfr = cwt_morlet(X, sfreq=128., freqs=[10.],  n_cycles=5, decim=2)
```

gets
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-c7cf187d065f> in <module>()
      2 from mne.time_frequency import cwt_morlet
      3 X = np.random.rand(10, 200)
----> 4 tfr = cwt_morlet(X, sfreq=128., freqs=[10.],  n_cycles=5, decim=2)

/home/jrking/mne-python/mne/time_frequency/tfr.py in cwt_morlet(X, sfreq, freqs, use_fft, n_cycles, zero_mean, decim)
    292     tfrs = np.empty((n_signals, len(freqs), n_times), dtype=np.complex)
    293     for k, tfr in enumerate(coefs):
--> 294         tfrs[k] = tfr
    295 
    296     return tfrs

ValueError: could not broadcast input array from shape (1,100) into shape (1,200)

```

I probably introduced the bug in my last PR, I'll have a look now

cmoutard, 2016-04-28:
I think this may be a bug since n_times is not updated given the decim argument.

cmoutard, 2016-04-28:
This should do the trick:
`n_signals, n_times = X.shape
decim = int(decim)
n_times = n_times  // decim + bool(n_times % decim)`

agramfort, 2016-04-28:
PR with test welcome

cmoutard, 2016-04-28:
Same bug in the function cwt

kingjr, 2016-04-28:
> Same bug in the function cwt

Are you sure? This seems to work:

```
import numpy as np
from mne.time_frequency.tfr import cwt, morlet
X = np.random.rand(10, 200)
# Precompute wavelets for given frequency range to save time
Ws = morlet(128, [10.])
coefs = cwt(X, Ws, decim=2)
```

cmoutard, 2016-04-28:
It does work, my bad

closed 2016-04-28 by Eric89GXL