mne-tools/mne-python #3189: Q: makes EpochsArray events defaults to np.zeros((len(data), 3), int)

kingjr, 2016-04-28:
Is there a reason to not default the `events` parameter of `EpochsArray` to `np.zeros((len(data), 3), int)`?

Related to this, I often find the creation of `Epochs` to be an overkill: i.e. I often end up doing
```
from mne import create_info
info = create_info(data.shape[1], sfreq=256., ch_types='mag')
epochs = EpochsArray(data=data, info=info, events=np.zeros((len(data), 3), int)
```

where I would expect only doing:
`epochs = EpochsArray(data, sfreq=256)`

Eric89GXL, 2016-04-28:
What channel type should you get for that one liner? (I'm -1 on that bit
because we want people to think about and be explicit about channel types.)

For events, sure we could do it but zeros isn't good enough. We'd probably
want ones in the last column and np.arange in the first.

kingjr, 2016-04-28:
>  For events, sure we could do it but zeros isn't good enough. We'd probably want ones in the last column and np.arange in the first.

+1

> What channel type should you get for that one liner? (I'm -1 on that bit because we want people to think about and be explicit about channel types.)

I was thinking of putting `mag` by default, but fair enough. The point is that we need to call 2 objects to do this basic operation.

However, I'm not sure what to propose instead without inducing deprecation or multiple ways to do the same things.

agramfort, 2016-04-28:
I am fine with an automatic events but that's it.
users should be aware of the info and know what they are doing.

closed 2016-05-01 by kingjr