mne-tools/mne-python #3200: ENH: plot_joint_psd

jona-sassenhagen, 2016-04-30:
Do we want this?

[image]

kingjr, 2016-04-30:
If you find a better example, maybe?

kingjr, 2016-04-30:
Actually the plot_joint with the topos can be good, although I would normalize per window to make it easier to read and avoid the 1/f non-informative gradient, but the colored line are not super convincing (although for consistency, why not)

agramfort, 2016-05-01:
if you do it on somato with proper baselining it could be nice.

dengemann, 2016-05-01:
It could be nice but not with the PSD default settings that we have, for
low frequencies we need long windows, and the welch periodograms don't look
nice. Detrending, power normalization and loglog plotting are further
options. I have some private code on this but I'm hesitating to push it
here as it is pretty particular.

dengemann, 2016-05-01:
second, the joint plotting needs a rewrite, on resizing the series and the
topo plots get misaligned. very often you want the series plot to be bigger
in this case.

jona-sassenhagen, 2016-05-01:
Okay, let's see ... also I have way too many issues open already :)

closed 2016-05-01 by jona-sassenhagen

dengemann, 2016-05-01:
Let's get back to it in a while once we discussed and solved the issues
mentioned in our use cases.

dengemann, 2016-05-01:
And a related issue I'm holding back is of course a PSDEpochs container
with .freqs instead of .times as sample descriptor axes.