mne-tools/mne-python #3214: FIX: some missing .copy

dengemann, 2016-05-05:
```
Users/dengemann/anaconda/envs/mnedev/lib/python2.7/site-packages/ipykernel/__main__.py:2: DeprecationWarning: The copy parameter is deprecated and will be removed in 0.14. In 0.13 the behavior will be to operate in-place (like copy=False). In 0.12 the default is copy=True. Use inst.copy() if necessary.
  from ipykernel import kernelapp as app
```

on calling Label.morph

cc @Eric89GXL

Eric89GXL, 2016-05-06:
good call, will fix tomorrow

Eric89GXL, 2016-05-06:
`Label` already has a `copy` method:

https://github.com/mne-tools/mne-python/blob/master/mne/label.py#L388

Did you mean something else?

closed 2016-05-06 by Eric89GXL

dengemann, 2016-05-06:
Not sure downstream some warning is caused ... could not identify it at one glance.