mne-tools/mne-python #3221: maxwell_filter not working with Triux files - channel order issue

Mountainmums, 2016-05-11:
I've been trying to Maxwell filter Triux fif files and have been encountering issues related to the channel ordering. Because Triux files put EOG etc data before MEG, `_update_sensor_geometry` fails on line 1392. 
 
```
    grad_picks = pick_types(info, meg='grad', exclude=[])
    mag_picks = pick_types(info, meg='mag', exclude=[])
    grad_imbalances = np.array([cal_chs[ii]['calib_coeff']
                                for ii in grad_picks]).T
```

The channel numbers returned by grad-picks do not align with the size of `cal_chs` which lead to an out of range index error. 
Fixing this problem with a dirty hack did not suffice, as further matrix size mismatches appeared in `_sss_basis_point` at line 1427:
 `       S_add *= imb[:, np.newaxis]`

I am not sure how to properly fix this.

Eric89GXL, 2016-05-11:
Can you provide a TRIUX file plus your cal file for us to use and add to our testing data to fix it? If you have an empty-room file (i.e., anonymized already) that would be great. We can take care of trimming it down to a small size.

Mountainmums, 2016-05-12:
I uploaded a Triux ER file [https://www.dropbox.com/s/ktigl4kov1w7zxl/Triux_ER.fif?dl=0](url)  as well as both ct_sparse [https://www.dropbox.com/s/3nw27ccxa4mrm9o/ct_sparse_triux.fif?dl=0](url) and sss_cal [https://www.dropbox.com/s/7l42tz3bin3pkd5/sss_cal_triux.dat?dl=0](url) files to dropbox.

Eric89GXL, 2016-05-12:
Great, thanks. Is it okay if we add that file (or some modified version of it) to our testing dataset? Also, which site is this from?

Mountainmums, 2016-05-12:
No problem with adding the file to the testing dataset. The system is in Helsinki, Biomag laboratory in Helsinki Meilahti Hospital.

Eric89GXL, 2016-05-13:
@Mountainmums the latest version of `master` has been fixed. Feel free to give it a try when you get a chance.

closed 2016-05-13 by Eric89GXL

Mountainmums, 2016-05-16:
Downloaded and tried the `master `code today, but still can't get it to work. It only works if I don't use the cross-talk or fine calibration options.

To check what I was doing wrong, I downloaded the test data and ran the test code, but I could not get that to work either.

The filtering with no calibration files (` sss_py = maxwell_filter(raw, coord_frame='meg', regularize=None)` works. 

Adding the cross-talk ` sss_py = maxwell_filter(raw, coord_frame='meg', regularize=None,
                            cross_talk=tri_ctc_fname)`  results in a `dimension mismatch` error in `maxwell_filter` line 406.

Adding the fine calibration results in a `list index out of range` in  `_update_sensor_geometry ` line 1705. 

Is there something funky that could be going on in my setup ?

Eric89GXL, 2016-05-16:
What is mne.__version__? I suspect you might still be on the release
version, not master.

Eric

Mountainmums, 2016-05-16:
mne-0.13.dev0-py2.7.

Eric89GXL, 2016-05-16:
Can you paste the trace back and a minimal script that fails on the test
data?

Eric89GXL, 2016-05-16:
And can you give the output of mne.sys_info()?

Mountainmums, 2016-05-16:
> ne.sys_info()
> Platform:      Windows-2008ServerR2-6.1.7601-SP1
> Python:        2.7.11 |Anaconda 2.5.0 (64-bit)| (default, Jan 29 2016, 14:26:21) [MSC v.1500 64 bit (AMD64)]
> Executable:    D:\Anaconda2\python.exe
> 
> mne:           0.13.dev0
> numpy:         1.10.4 {lapack=mkl_lapack95_lp64, blas=mkl_core_dll}
> scipy:         0.17.0
> matplotlib:    1.5.1
> 
> sklearn:       0.17
> nibabel:       Not found
> nitime:        Not found
> mayavi:        Not found
> nose:          1.3.7
> pandas:        0.17.1
> pycuda:        Not found
> skcuda:        Not found

Eric89GXL, 2016-05-16:
That version you gave must come from pip, but what I'm looking for its the
version from within python as `mne.__version__`

Mountainmums, 2016-05-16:
I am just running the script from test_triux()

`    sss_py = maxwell_filter(raw, coord_frame='meg', regularize=None,
                            cross_talk=tri_ctc_fname)`
¨Traceback (most recent call last):

> 
>   File "<ipython-input-86-92d028cf4075>", line 2, in <module>
>     cross_talk=tri_ctc_fname)
> 
>   File "<string>", line 2, in maxwell_filter
> 
>   File "D:\Anaconda2\lib\site-packages\mne-0.13.dev0-py2.7.egg\mne\utils.py", line 629, in verbose
>     return function(*args, **kwargs)
> 
>   File "D:\Anaconda2\lib\site-packages\mne-0.13.dev0-py2.7.egg\mne\preprocessing\maxwell.py", line 406, in maxwell_filter
>     t_str += ('(#%d/%d)'
> 
>   File "D:\Anaconda2\lib\site-packages\scipy\sparse\base.py", line 253, in dot
>     return self * other
> 
>   File "D:\Anaconda2\lib\site-packages\scipy\sparse\base.py", line 355, in __mul__
>     raise ValueError('dimension mismatch')
> 
> ValueError: dimension mismatch

```
    sss_py = maxwell_filter(raw, coord_frame='meg', regularize=None,
                            calibration=tri_cal_fname)
```

> Traceback (most recent call last):
> 
>   File "<ipython-input-87-5b13ff77b720>", line 2, in <module>
>     calibration=tri_cal_fname)
> 
>   File "<string>", line 2, in maxwell_filter
> 
>   File "D:\Anaconda2\lib\site-packages\mne-0.13.dev0-py2.7.egg\mne\utils.py", line 629, in verbose
>     return function(*args, **kwargs)
> 
>   File "D:\Anaconda2\lib\site-packages\mne-0.13.dev0-py2.7.egg\mne\preprocessing\maxwell.py", line 292, in maxwell_filter
>     calibration, sss_cal = _update_sensor_geometry(info, calibration,
> 
>   File "D:\Anaconda2\lib\site-packages\mne-0.13.dev0-py2.7.egg\mne\preprocessing\maxwell.py", line 1705, in _update_sensor_geometry
>     adjust_logged = True
> 
> IndexError: list index out of range
>

Eric89GXL, 2016-05-16:
There is something very strange going on, since the first one complains about a `dot` on a line that does no dot product, and the second one is a simple `adjust_logged = True` statement, but it gets an `IndexError`. I think this must be a problem with the installation of latest `master`, it somehow failed and got in a bad state. How did you upgrade? Can you try removing `mne`, ensure it doesn't import, and tehn reinstall latest `master`?

Mountainmums, 2016-05-16:
Yeah, removing everything and reinstalling the latest `master `did the trick. Sorry for the hassle.

Eric89GXL, 2016-05-16:
Thanks for sticking with it. Probably not your fault actually but some
Python installation bug. What was the command you originally used to update
to latest master? I want to try to reproduce the sequence locally and
update our docs if necessary.

Mountainmums, 2016-05-16:
I actually installed it manually, couldn't somehow get the pip command to do the install in the Anaconda directory. Dowloaded the master zip, copied the unzipped mne dir into D:\Anaconda2\Lib\site-packages\mne  and ran `python setup.py install` from the command line, which created the additional D:\Anaconda2\Lib\site-packages\mne-0.13.dev0-py2.7.egg directory.

Eric89GXL, 2016-05-16:
Do you remember what error you were getting with pip? It would be best if
that worked properly for people.

Mountainmums, 2016-05-16:
I didn't get any error message per say. It's more likely that it's my lack
of python knowledge that's the real probem here.

The pip command installed the toolbox in a src directory on my C/: user
directory rather than in my Anaconda root. Since I wasn't sure what would
happen if I ran the setup from there (maybe I should have tried), I just
went the manual way for the install.

Eric89GXL, 2016-05-16:
> The pip command installed the toolbox in a src directory on my C/: 

Which command did you run exactly, `pip install -e git+https://github.com/mne-tools/mne-python#egg=mne-dev --user`?

Eric89GXL, 2016-05-16:
And if you still have a log or can recreate it, knowing exactly where it installed it would be useful.

Mountainmums, 2016-05-16:
Yeah, that was the command I used. I'll run it again tomorrow and post the
log and info.

-- 
-- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - --
Muriel LOBIER
Post-doctoral Scientist
Neuroscience Center <http://www.helsinki.fi/neurosci/groups/palva_satu.html>
- University of Helsinki
<muriel.lobier@upmf-grenoble.fr>
tel : +358 (0) 44 0347333
Web : www.aconnectedbrain.com
<http://webu2.upmf-grenoble.fr/LPNC/membre_muriel_lobier>

Eric89GXL, 2016-05-16:
Great, thanks. If you could also do the same command without the `--user` appended that would also help. I think our README is out of date and the `--user` is not supposed to be there, which might be what led you astray.

Mountainmums, 2016-05-16:
Think I tried both options, but I'll check tomorrow.

-- 
-- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - -- - - --
Muriel LOBIER
Post-doctoral Scientist
Neuroscience Center <http://www.helsinki.fi/neurosci/groups/palva_satu.html>
- University of Helsinki
<muriel.lobier@upmf-grenoble.fr>
tel : +358 (0) 44 0347333
Web : www.aconnectedbrain.com
<http://webu2.upmf-grenoble.fr/LPNC/membre_muriel_lobier>

Mountainmums, 2016-05-17:
So I played around with things a bit and figured out what had me confused. Running` pip install -e git+https://github.com/mne-tools/mne-python#egg=mne-dev` with or without `--user `behaves similarly for me: it downloads mne to a /src dir created in whichever dir I am running the command from. 
```
Obtaining mne from git+https://github.com/mne-tools/mne-python#egg=mne-dev
  Updating d:\anaconda2\src\mne clone
Installing collected packages: mne
  Running setup.py develop for mne
Successfully installed mne-0.13.dev0
```
What messed me up is that I didn't know that the -e option leads pip to not install in /version-packages but in a different src folder. 
Knowing that it seems to be working ok.