mne-tools/mne-python #3232: Epoching problem-CNT data

cfoodc, 2016-05-13:
Hi all,

I encountered a problem on my CNT data.
After epoching, I found that data in several channels became all the same.
(in this case, channel index 1,3,5,7,11,13,15....etc, their values are identical)
I am not sure what's going on.

The ipynb and original CNT were attached.
It might be easier to reproduce the problem from the actual files. 
https://www.dropbox.com/s/gnbur10vizbsfqw/bug.zip?dl=0

Best,

CF

cfoodc, 2016-05-13:
I just found there is a same problem during the raw data stage.
Several channels are just flat lines.

[image]

agramfort, 2016-05-13:
so is it a software problem or just a data problem? If so please close.
thx

cfoodc, 2016-05-13:
I have tried to read the data with spm and eeglab, the data looks like normal.
So it might be a software problem.

Eric89GXL, 2016-05-13:
@jaeilepp do you have time to look at this one?

jaeilepp, 2016-05-17:
Taking a look now.

jaeilepp, 2016-05-17:
Looks like the problem is that the data is in 32 bits instead of 16. The annoying thing is that this info is not included in the header, so I'll have to infer it from the data.

closed 2016-05-21 by agramfort