mne-tools/mne-python #3239: ENH: add `auto` param in filter functions

kingjr, 2016-05-18:
Currently, classic high pass filtering (i.e. `< 0.2` Hz) crashes when uses default, because the transition bandwidth are too large.

As changing the default would create backward incompatibility and differences with MNE-C .

We suggest to add an `auto` param to adapt the transition bandwidth according to the frequency. 

cc @choldgraf @Eric89GXL @jona-sassenhagen

Eric89GXL, 2016-05-19:
From talking to @rkmaddox, our defaults might be a little extreme. For example, a 10-second FIR with a 0.5 Hz transition to low-pass at e.g. 40 Hz might be a bit overkill. At the very least, adding `length='auto'`, `*_trans_bandwidth='auto'` could alleviate this a bit, and making them the defaults after a couple of releases might be worth thinking about.

Eric89GXL, 2016-05-19:
I'll probably dig into this issue at the same time as #3227 (filtering FAQ), hopefully this summer sometime.

I should also look and see what MNE-C does for filtering, since I don't think it's identical to what we do (IIRC our filters are at least longer).

agramfort, 2016-05-19:
I am fine with auto options if they are well tested before merging

choldgraf, 2016-05-19:
I also like the 'auto' idea, and am very +1 for a guide / tutorial / whatever on filtering. It's one of the issues that's commonly not well understood in neuroscience IME. Happy to give feedback on writing etc.

It would be awesome if there were enough MNE materials to basically fill a semi-advanced undergrad course in neuro data analysis :D

dengemann, 2016-05-19:
Better defaults would be nice here. Our filter API is a bit cumbersome.

Eric89GXL, 2016-05-23:
It just occurred to me that we could also give some warnings when people use filters that are too long for the given data, e.g. a FIR filter with too many coefficients or an IIR filter that rings too long (although this might be too expensive to estimate, I'd have to see).

choldgraf, 2016-05-23:
I'd be +1 for that if it doesn't take too much time, though I bet using an
'auto' option as well as including some good documentation on this would be
sufficient.

PR #3449 (2016-07-29): [MRG+1]: Add filtering default
Closes #3239.

Still need to update the background tutorial a bit, plus update `whats_new.py`.

Sorry in advance for the big diff...
closed 2016-07-29 by agramfort