mne-tools/mne-python #3242: BUG: DigMontage and related

kingjr, 2016-05-20:
`DigMontage` doesn't check the input types properly.

E.g. `dev_head_t` is not checked to be a 4x4 np.array (I passed a `mne.transform.Transform` instead).

Related to this issue, when doing `epochs.set_montage(DigMontage(...)).save()` this starts to save the epochs, but crashes because of the wrong montage. This means that if you overwrite a file, you lose the data.

I'm thus wondering whether we shouldn't have safer saving function that first save in a temp file and change name once we know no further error can be done.

kingjr, 2016-05-20:
To be more explicit you can do 

` DigMontage('bar', 'foo', 'foo', 'foo', 'foo', 'foo', 'foo', 'foo', 'foo')`

without any error.

agramfort, 2016-05-21:
can you give a script to replicate the bug?

kingjr, 2016-05-21:
```
from mne.io import Raw
from mne.datasets import sample
from mne.channels import DigMontage
fname = sample.data_path() + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = Raw(fname)

# Passing wrong values to Digmontage never crashes, but inst.set_montage would:
montage = DigMontage(0, 0, 0, 0, 0, 0, 0, 0, 0)  # should crash & doesn't
raw.set_montage(montage)  # should crash and does.

# Some variable in montage are not checked either during DigMontage or
# set_montage:
montage = DigMontage(hsp=[[0, 0, 0]],
                     hpi=[[0, 0, 0]],
                     elp='foo',
                     point_names='foo',
                     nasion=[0, 0, 0],
                     lpa=[0, 0, 0],
                     rpa=[0, 0, 0],
                     dev_head_t='foo')
raw.set_montage(montage)  # should at least crash

# this creates save error
raw.save('test_raw.fif')

# if you overwrote your 'test_raw.fif': the saving has started (file rewritten)
#  but crashed in the middle. I thus suggest we systematically write a file
# in '.*.tmp' and move its content at the end (like what is down for downloads
# in your favorite browser)
```

agramfort, 2016-05-23:
@kingjr did you loose data? this tmp file moving seems a bit an overkill
... let's rather fix the IO pb with a better check_info that crashed before
starting to write

kingjr, 2016-05-23:
I did but I shouldn't have overwrote on the same filename, so it's partly my fault. (I still have backups so no worry)

The temp moving proposition just a generic caution idea, but I agree the priority is to fix the DigMontage.

Eric89GXL, 2016-11-18:
This has (more or less) been addressed via documentation, `DigMontage` isn't meant to be instantiated by the user but rather through `read_dig_montage`

closed 2016-11-18 by Eric89GXL