mne-tools/mne-python #3243: Q: CTF coregistration with separate coil position?

kingjr, 2016-05-20:
### Problem

I am trying to coregister CTF data with an MRI to produce a `-trans.fif`

### What I have

The MEG data is separate from the positions in digitizer space so the `raw` file doesn't know about: `hpi_coils.txt`, `head_shape.txt`, `fiducials.txt`.

The HPI coil positions in MEG device space seem to be stored in the `raw` file: `'dig_points'` and `'coord_trans'` are present in `raw.info['hpi_results']` after reading the raw.

### Where I was going

I thought I would

1. create a `mne.channels.DigMontage` from this data (I am just unsure whether `raw.info['hpi_results'][0]['coord_trans']` corresponds to the parameter `'dev_head_t'` required by `DigMontage`).

2. apply this montage to my raw file using `raw.set_montage(my_new_montage).save(raw_fname)`.

### Where I'm stuck

However, `mne.gui.coregistration` still does not show the head shape positions nor the HPI coils.

Any idea how to proceed?

(For the KIT, thanks to @teonbrooks, this is handle at the reading stage (`read_raw_kit` needs the headshape to create the `raw`), and in the Elekta the position are in the raw.fif in the first place.)

agramfort, 2016-05-21:
no idea without inspecting the data

teonbrooks, 2016-05-22:
hey, just now seeing this, the approach is that your digitized data (headshape (in head space), head position indicator points (in MEG space), electrode location positions (in head shape)) should be used to generate a DigMontage (we can have a discussion if this nomenclature is unclear or confusing, and possibly have a clearer use case if it doesn't make sense). 
For the coil position, assuming hpi_coils correspond to the MEG sensor coil positions, you can generate a hpts file (it's specified in the mne manual) and these sensors will be properly added. 

are these files you have for the CTF standard? would it be better to expand the `read_raw_ctf` so that it handles these cases?
also, we can add a new page to the manual for ctf data if we resolve this to document the procedure

kingjr, 2016-05-22:
Thanks @teonbrooks. I'll try to check the hpts file generation in the manual next week, and eventually upload the data if I remain stuck.

I don't think the files coding for the head shape etc are standard. However, IIUC, CTF doesn't naturally include head shape etc in the .ds folder, so we may need to add a tutorial on this specific point.

romquentin, 2016-05-27:
Hey, 
Thanks for the help. 
Here is a link to the data (https://www.dropbox.com/s/a7lx0oe264kr9f8/CTF_source.zip?dl=0) with:

1- MRI T1
2- epochs.fif
3- headshape.txt (with 262 points and fiducials at the end)
4- HPIcoils.txt 

We used Brainsight (a set-up with subject tracker and infrared camera mainly use for neuronavigation during TMS stimulation) to create headshape.txt and HPIcoils.txt. The coordinates in the headshape.txt are relative to the subject tracker and HPIcoils.txt use the coordinate system based on anatomical scan. Unfortunately direct comparisons are not possible. I also have photo of HPIcoils on the head participant and I think I can position manually the HPIcoils on the subject MRI using mne_gui_coregistration?

I tried to run `mne_ctf_dig2fiff --dig headshape.txt --fif headfif --hpts headhpts`and the error is:

`262 digitization points in headshape.txt
Illegal or missing data in position file : <28.432218 -61.507432 -77.752731
`
Thanks again!

Romain

agramfort, 2016-05-29:
is someone on this?

jaeilepp, 2016-05-30:
I believe the problem with the conversion is in the format of the data. It seems that there needs to be an ID for each digitization point. So I tried adding a number in front of couple of the points and it seemed to work.

romquentin, 2016-06-01:
Thanks. Indeed, it is working with the number before the 3 coordinates.

romquentin, 2016-06-01:
But I do not know how to use the hpts file now. Should I use mne.gui.coregistration?
I still have the HPIcoils.txt in the anatomical MRI coordinates in mm. 
In general, should it be better to have all informations (fiducials, headshape and HPIcoils position) in the anatomical MRI space?
Thanks a lot!

agramfort, 2016-06-01:
use mne_analyze if you have it working (mac or linux).

see:
http://martinos.org/mne/dev/manual/gui/analyze.html#coordinate-frame-alignment

romquentin, 2016-06-01:
Ok. Thanks. I will try that (I am on mac).

romquentin, 2016-06-02:
When I run mne_analyze, I get this error:
`dyld: Library not loaded: libquicktime.dylib`
 `Referenced from: /Applications/MNE-2.7.4-3420-MacOSX-x86_64/bin/mne_analyze`
 `Reason: image not found`
`Trace/BPT trap: 5
`
Any ideas?
Thanks

agramfort, 2016-06-02:
how did you setup mne-c?

romquentin, 2016-06-02:
Thanks... I changed the .bash_profile and it works!

romquentin, 2016-06-06:
With mne_analyze, I can open my digitized file 'headshape.fif' and the surfaces files and use the 'adjust coordinate alignment' to save the '-trans.fif'. 
But I have my HPI coordinates (in the MRI coordinates) and I do not know how to use this information. Does the fiducial that I can mark manually in mni.gui.coregistration corresponds HPI position?
And more generally, If I have my HPI position in the MRI coordinates, do I really need a digitized headshape?
Thanks a lot for your help!
Romain

agramfort, 2016-06-07:
> With mne_analyze, I can open my digitized file 'headshape.fif' and the
> surfaces files and use the 'adjust coordinate alignment' to save the
> '-trans.fif'.
> But I have my HPI coordinates (in the MRI [...]
not HPI but the first 3 landmarks in the dig usually.

> And more generally, If I have my HPI position in the MRI coordinates, do I
> really need a digitized headshape?
> Thanks a lot for your help!
>
the 3 fiducials are mathematically enough, yet it's highly recommended to
use as many digitized points to have a good coreg.

romquentin, 2016-06-07:
Sorry, it seems I am confused between fiducials and HPI positions. Where do I need to enter the information about my HPI coils position? (which is in my case different from the fiducials in my digitized headshape). Thanks!!

dgwakeman, 2016-06-07:
@romquentin fiducials and hpi positions are always different. you need to match up the fiducials to the mri this then provides a transformation matrix which can translate all other positions to MRI space e.g. HPI coils and extra points (eeg if present). If this wasn't clear from the documentation, please consider a PR to help clarify our documentation. hth d

romquentin, 2016-06-07:
Ok. Thank you. I would be happy to help clarify the documentation. I did not find this information.
If I understand correctly, as I already have the HPI coordinates in MRI space (from a neuronavigation system called Brainsight), I do not need to transform these coordinates? My question is how I can use this information (HPI coordinates in MRI space) in mne.gui.coregistration? Hope it is not confusing. Thank you

agramfort, 2016-06-07:
what do you see once loading the dig points in coreg GUI?

romquentin, 2016-06-08:
With `mne_analyze` : I can load surface and digitalized headshape data (.fif obtained from mne_ctf_dig2fiff). Then adjust coordinate alignment --> ICP align and I can save the `-trans.fif` 

With `mne.gui.coregistration` : I can also load the surface and specify manually the fiducials. But when I try to open the digitalized headshape data from "head shape source". The return error is : `ValueError: Could not find measurement data` and I can not see the head shape points.

And finally, I still don't know how I can use my HPI coils coordinates in MRI nifti space. The transformation (-trans.fif) is useless because these coordinates are already in a MRI space I guess. But I did not find how to use this information. Let say I want to specify manually where was the HPI coils from pictures I took before entering the MEG. Where can I enter this information?

Thanks a lot for your help!!

dgwakeman, 2016-06-08:
@romquentin, Well you can use the one from mne_analyze. That should have you all set.

for the python one, I'm sure one of the experts here can help you get that to work.

Using the one you did in other software would require significant work. It may not be worthwhile to create all of the necessary custom code to translate it. You would need at least 2 additional transformations to make it work and then there would be several other steps necessary for adapting it. This would be a significant undertaking to code all of this.

romquentin, 2016-06-08:
Ok. Thanks a lot. But I guess at the end the important information is: where are the HPI coils on the MRI brain of the participant? I still do not see at what step this information is given to mne.

dgwakeman, 2016-06-08:
You should see them in mne_analyze. They are green there. Hth d

romquentin, 2016-06-08:
I have one .txt file with fiducials and head shape and another .txt file with HPI coils position (in MRI space). Coordinate system in these two .txt files are not similar. I converted the .txt file containing fiducials and headshape to a .fif with mne_ctf_dig2fiff function and I loaded this .fif in mne_analyze. From that I saved a -trans.fif. 
But I did not load the .txt file with HPI coils position in mne_analyze so there is no chance that I can see them. How can I load this information in mne_analyze? Maybe the HPI coils coordinates should be in the same text file than fiducials and head shape at the beginning? And with the same system coordinate.

dgwakeman, 2016-06-08:
ok, you should convert the files and add the raw (which you haven't registered in some other program) digitization information first to the fif files then use mne_analyze to do the co-registration. There is extensive documentation in the mne-c manual evoked with mne_view_manual in section 9.2.2 Importing CTF data. Follow those then start again from mne_analyze and the file will have everything.

kingjr, 2016-06-08:
This is getting very difficult to follow let's recap @romquentin

Correct me if I'm wrong, but from what I understand, you need to

- [x] 1. Ensure you have a working MNE C (with `.bashrc`)
- [x] 2. Modify your `headshape.txt` file so that it contains an `ID` for each digitized point.
- [x] 3. Generate a `hpts` file with the MNE C command `mne_ctf_dig2fiff`. This will give the positions of the HPI coils in MEG/sensor space.
- [ ] **Missing step**: manually indicate where the HPI coils are located in head space (this is why you have a picture). @teonlamont, do you know how he can do this?
- [x] 4. Launch the MNE C command `mne_analyze`
- [ ] 5. Manually indicate your fiducials (i.e. in head space) by clicking on LPA, nasion, & RPA.
- [ ] 6. Load your head digitized points (i.e. in head space) by clicking on `file>load Digitizer data`
- [ ] 7. Fit (`Adjust>coordinates aligment`) your fiducials and/or digitized points so that you know how to align the head space to the MEG space. ( 3 coordinates is enough in principle, but the more points you have, the more reliable will be your realignment between head space and MEG space).
- [ ] 8. Save a `-trans.fif` file.
- [ ] 9. Make your forward and inverse model.

romquentin, 2016-06-08:
Thank you @dgwakeman and @kingjr .

1. For the missing step, it is ok to indicate manually the HPI coils. Nevertheless, I have the HPI coils coordinates (in MRI space but I guess I can transform them to the same coordinate system than the one in headshape.txt) in a separate .txt file. It should be more precise to use these coordinates if possible.

2.  I can skip step 5 I guess because my fiducials are included in my head digitized points.

kingjr, 2016-06-08:
> I can skip step 5 I guess because my fiducials are included in my head digitized points.

No, do this step too. The digitized points on the head tend to be spherical, and you can therefore tilt the whole alignment. Fitting the MRI (manual/clicking) fiducials with the one you digitized (in head space) prevents this issue.

For the **missing step**, let's wait for @teonbrooks input.

kingjr, 2016-06-10:
@lauragwilliams could you share the code you wrote to integrate the HPI in the raw fif if you think this could solve the issue?

teonbrooks, 2016-06-10:
I think there should be two approaches: short-term solution, I will meet up with @kingjr IRL and we will write a gist for computing the `dev_head_t` using the existing functions in MNE. long-term solution, there needs to be a `read_raw_ctf` module written in Python to get it up to speed with the other systems.

LauraGwilliams, 2016-06-10:
I've uploaded the code here: https://gist.github.com/LauraGwilliams/41d1a543c10dec181b02f4e6ac659fde#file-get_dev_t-py. The relevant stuff is from line 70. Let me know if something isn't clear.

kingjr, 2016-06-12:
Thanks @LauraGwilliams !

@romquentin: @teonbrooks and I met and manage to incorporate the HPI, fiducials and digitized head shape points in an MNE object: https://gist.github.com/kingjr/8f971c73c79712b547cdd55c661c6a5a

you need to : 
* write a parser to extract the HPI, head shape and fiducials as numpy arrays similar to what is in the gist.
* duplicate your HPI, as the `read_dig_montage` expect hpi points that are called `lpa`, `rpa` and `nasion` for the coordinates transformation, and will use the other as hpi coils.
* save your mne object on the hard drive as `mne.gui.coregistration` expects a file and not an mne instance directly.

If you're ok, I'll close the issue.

romquentin, 2016-06-13:
Great! Thanks a lot! 
Does coordinate system need to be the same for all position (HPI, fiducial and headshape)?

kingjr, 2016-06-13:
HPI are in device space (MEG), fiducials and headshape are in head space.

romquentin, 2016-06-13:
Ok. Thanks again!

closed 2016-06-13 by kingjr

romquentin, 2016-10-06:
@kingjr: Where do you find the hpi information here: https://gist.github.com/kingjr/8f971c73c79712b547cdd55c661c6a5a
I thought it was in raw.info but it doesn't seem to be similar to the coordinate that you use.
Thank you!