mne-tools/mne-python #3247: FLASH bem pipeline

kambysese, 2016-05-25:
It looks like `mne flash_bem` has undergone significant overhaul since `0.11` I am having issues making the script play nice with existing  structural databases i.e. without using `mne_organize_dicom`. What is the expected input into `mne flash_bem`? When I invoke the script in a directory with files linked to the 5/30 flash files in nifti format, I get the following trace:

> Traceback (most recent call last):
>   File "/home/ktavabi/Projects/mne-python/mne/commands/mne_flash_bem.py", line 90, in <module>
>     run()
>   File "/home/ktavabi/Projects/mne-python/mne/commands/mne_flash_bem.py", line 84, in run
>     flash30=flash30, convert=convert, unwarp=unwarp)
>   File "<string>", line 2, in convert_flash_mris
>   File "/home/ktavabi/Projects/mne-python/mne/utils.py", line 629, in verbose
>     return function(*args, **kwargs)
>   File "/home/ktavabi/Projects/mne-python/mne/bem.py", line 1569, in convert_flash_mris
>     % op.join('flash' + flash, echo))
> RuntimeError: Directory flash05/002 is missing.

On a related note `mne_organize_dicom` in a directory of dicom files throws a syntax error at line 23, which doesn't really make much sense.
> MNE-2.7.4-3505-Linux-x86_64/bin/mne_organize_dicom: Syntax error: "(" unexpected.

Eric89GXL, 2016-05-25:
If you comment out the bit about `mne_origanize_dicom` does it work? If so maybe we can add an option to at least `mne.bem.make_flash_bem`.

kambysese, 2016-05-26:
There is a `--noconvert` kwarg for `converts_flash_mris` which will work once I figure out if my flash files are properly converted and where they should be. Will close for now.

closed 2016-05-26 by kambysese