mne-tools/mne-python #3249: bug with multitaper DICS

agramfort, 2016-05-26:
see mailing list message:

```
In order to compute the single-trial power at the source level at a single frequency, the steps I normally use (in Fieldtrip) are:
1) Multitaper at the sensor level: compute complex-valued estimates of spectral measures Xsensor(n, t, k) , for each trial n, time t, and taper k. The cross-spectral density matrices, however, is computed across trials (average csd).
2) DICS: compute complex-value spectral measures at the source level, Xsource(s, t, k) =  A * Xsensor(n, t, k) , where A(t) is the real-valued spatial filter that transforms the data from the sensor to source level, which is computed from the real part of the csd in (1). The Xsource(s, t, k) is calculated for each source s, time t, and taper k
3) The single-trial power at each source location is computed by multiplying the complex spectral estimates with their complex conjugate, and averaged over tapers k. Psource(s, t) = <   Xsource dot Xsource(s, t, k) * > over k
4) Finally, I log-transform the power values to make the data approximate Gaussian and lowpass filtered at 50 Hz to reduce noise. Then, the single-trial mean power and SD in the baseline period (i.e., multiple estimates of noise) are computed for each source and trial, and used to z-transform single-trial event-related power time courses.

I hope this is clear. If you want a detailed description you can read the paragraph "Single-trial HGA at BAs" in the Methods in this paper:
http://www.jneurosci.org/content/35/37/12643.short

However, I am not familiar with MNE yet, I am slowly learning. So, I don't know exactly how I could contribute to implement this. But I can help ! Just let me know.
```

brovelli, 2017-07-18:
Dear all,
I have a new def for dics.py that can be used to compute single-trial power at the source level using DICS as specified in the previous comment.
It could be called **def dics_source_power_epochs**
Can you tell me how to proceed? Shall I make a PR on dics.py?
Thanks

Eric89GXL, 2017-07-18:
@britta-wstnr any thoughts?

agramfort, 2017-07-18:
@brovelli if you've never done a PR you should start with a gist you can share?

brovelli, 2017-07-19:
For now I have it on my fork of mne-python
https://github.com/brovelli/mne-python/blob/master/mne/beamformer/_dics.py

agramfort, 2017-07-19:
thanks for sharing

@britta-wstnr <https://github.com/britta-wstnr> can you have a look? does
it overlap with your refactoring work?

britta-wstnr, 2017-07-19:
I think this issue will profit from the refactoring into ``make_dics`` (create the filter) and ``apply_dics`` (apply filter to data). It should make it much easier to have single trial estimates returned.

brovelli, 2017-07-19:
NB: my implementation does not use filters computed on single trials, but rather on **all trials**. Then the filter computed on the averaged csd is used to map single-trial data from sensors to source level.

I am not familiar with such refactoring change. Can you point me to such discussion or issue? thanks

brovelli, 2017-07-19:
And, in order to use multitaper method, the csd from single tapers must be kept all along.

britta-wstnr, 2017-07-19:
Yes, the filter would be calculated on the CSD-matrix of all trials. It is basically a change in the code which makes it possible to have the computed filter returned. It can then be applied to, e.g., single trials or the data of one conditions.
We are currently implementing this for the LCMV beamformer (see here for the PR: https://github.com/mne-tools/mne-python/pull/4389 ) and the plan is to also implement it for the DICS beamformer.

brovelli, 2017-07-19:
Great!