mne-tools/mne-python #3252: issues with loaded ICA object

sstober, 2016-05-26:
Hello,
I've stumbled across several issues working with a loaded ICA object. This object was created with 
```
ica = ICA(n_components=1.0, method='extended-infomax')
ica.fit(epochs) 
```
From the log messages, it looks like it is properly computed:
> [...] 
> computing Extended Infomax ICA

The object is then saved with ica.save(...) and later loaded with read_ica(...).

When I print the loaded object it says:
> <ICA  |  epochs decomposition, fit (fastica) [...]

So it looks like the fit was either not stored or read correctly.

Another issue arises, when I call ica.find_bads_eog(...). I get an error in [line 1048](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L1048) because ica.labels_ is None. It's passing the hasattr check in [line 1036](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L1036) thus is not properly initialized. Could be caused by read_ica(). My workaround is to set ica.labels_ = dict() manually. It would probably be a good idea to extend the check in [line 1036](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L1036) and [line 966](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L966) to include None.

Cheers,
Sebastian

agramfort, 2016-05-26:
@sstober thanks for the bug report.

@jaeilepp want to have a look?

agramfort, 2016-05-26:
@sstober can you give a try to https://github.com/mne-tools/mne-python/pull/3254 to see if it fixes your pb?

sstober, 2016-05-27:
Yes, the issues are resolved for newly computed ICAs. Thanks a lot! The previously saved ICA object computed with extended-infomax still loads as fastica though. I guess there’s no way to fix this.

jaeilepp, 2016-05-27:
Yeah, the value wasn't saved to the file before the fix. On read it defaults to 'fastica'.

closed 2016-05-27 by agramfort