mne-tools/mne-python #3262: Missing sphinx extensions

ARudiuk, 2016-05-27:
To get my make working I had to install sphinx-gallery and sphinx_bootstrap_theme. I don't see a reference to them in the contributing documentation, but it seems they should be under the list of development related tools.
Should I add them to the list of python libraries?

Eric89GXL, 2016-05-27:
Yes please add a section to contributing about how to build docs. It can start with these requirements and we can expand as you hit more snags :)

ARudiuk, 2016-05-27:
I did a couple things, and it seems to work now. The steps were essentially:
1]make cond up to date 'conda update --all'
2]check versions of libraries used in .travis.yml, 'PYTHON=2.7 DEPS=full TEST_LOCATION=src NUMPY="=1.9" SCIPY="=0.17"'
3]remove packages that are not working 'conda uninstall numpy scipy matplotlib mkl scikit-learn'
4]reinstall with constraints from step 2 'conda install numpy=1.9 scipy=0.17 matplotlib mkl scikit-learn'
5]at this step the make ran all the way through, but sklearn could not be imported, so I ran 'pip install sklearn --upgrade' and now everything seems to work

Eric89GXL, 2016-05-27:
Sorry for the pain, hopefully Anaconda can fix the dependency bugs soon. Don't include those steps in the contributor instructions, but maybe we should add those to the Python install instructions about what to do if Anaconda fails to resolve dependencies properly. Let's wait to do that (in a separate PR), though, in case Anaconda fixes their bug in the meantime.

ARudiuk, 2016-05-27:
So just the sphinx stuff for now?

Eric89GXL, 2016-05-27:
Yeah, a "how to build the doc" type section. If you have time, take a look at `doc/Makefile`, too -- it would be nice to document at least `make html_dev` and `make html_dev-noplot`, `make html_dev-pattern`.

ARudiuk, 2016-06-11:
I was looking at this today, and I was wondering how much detail you would want? The things to cover I can think of are:
-Required libraries
-What is built with each command

ARudiuk, 2016-06-11:
There also seems to be some outdated info like the command `make html`

agramfort, 2016-06-12:
sounds like a plan. Fix the pbs you spot

closed 2016-06-26 by agramfort

ARudiuk, 2016-06-26:
@agramfort  There are some things i was planning with this change that have not been addressed. Should I make a new issue?  
Specifically, a description of what each command does and removed references to old commands like `make html`

agramfort, 2016-06-26:
yes please do. Eric did a pass recently so start from a clean master

ARudiuk, 2016-06-26:
sounds good

PR #3337 (2016-06-26): MRG: Use numpydoc from pip
Gets rid of our `numpydoc` in favor of the one on `pip`. Also fixes a few problems with our docstrings.

Closes #3262.