mne-tools/mne-python #3264: Can't import data from EEG-LAB

pbrusco, 2016-05-27:
Hello, 

I'm having issues importing raw EEG data with mne.io.read_raw_eeglab(set_filename)

It seems to be related with the events on the file:

[image]

I tried importing the data by hand and seems to be working. 

[image]

Any ideas? 

mne.__version__
'0.12.0'

Thanks!

jona-sassenhagen, 2016-05-27:
How was the EEGLAB .set file created?

mmagnuski, 2016-05-27:
Some older eeglab files may have event type as integer. It is not allowed in current eeglab version (if I remember correctly). `eeg_checkset` (in matlab) should turn these to string.

jona-sassenhagen, 2016-05-27:
This could be easily fixed by forcing events to be str, but I'm not sure we should support old EEGLAB version files. Any opinions?

jasmainak, 2016-05-27:
is there a way to check what is the EEGLAB version from the `.set` file?

Eric89GXL, 2016-05-27:
Casting to string seems fine to me, assuming the older files otherwise work. But maybe we should emit a warning about old files if we can get the version as @jasmainak asked

jona-sassenhagen, 2016-05-27:
pbrusco can you make the file in question available somehow?

pbrusco, 2016-05-27:
Thanks for the fast answers (I think the files are a little bit old, but not too much)

https://dl.dropboxusercontent.com/u/43547597/test1.set
https://dl.dropboxusercontent.com/u/43547597/test1.fdt

jona-sassenhagen, 2016-05-27:
@jasmainak do you want to do it or should I?

jasmainak, 2016-05-28:
@jona-sassenhagen feel free to take a stab at it. Thanks!

pbrusco, 2016-06-01:
@jasmainak, in the mean time, do you recommend me to download the mne source code and just change cast events to string? Thanks :)

jona-sassenhagen, 2016-06-01:
Sorry, I'll try and get something done today.

jona-sassenhagen, 2016-06-01:
How do we go about testing for this? Probably don't want to add a new EEGLAB data file just for this.

@pbrusco in the meantime, I'd try loading and re-saving the file in a current version of EEGLAB.

pbrusco, 2016-06-02:
@jona-sassenhagen Ok, good idea ;), thanks!

jona-sassenhagen, 2016-06-03:
@pbrusco I think I don't understand your EEGLAB test file. It has only 1 trial?

mmagnuski, 2016-06-03:
pbrusco was using `mne.io.read_raw_eeglab` so it is probably a raw file.

jona-sassenhagen, 2016-06-03:
Haha I'm stupid.

closed 2016-06-06 by agramfort

pbrusco, 2016-06-06:
Worked perfectly! Thanks a lot

PR #3279 (2016-06-06): MRG robustness to old EEGLAB versions
Old EEGLAB versions seemingly allow integer event codes.
closes #3264

Not sure how to test - we probably don't want to add another test file just for this.