mne-tools/mne-python #3269: ENH: ICA component raster plots

espressofiend, 2016-05-29:
Especially for EEG, many ICs may be noise - we often see ~5-6 EOG components (including both blinks and horizontal movements), as well as noisy channels and other things. ica.plot_scores does not detect all of these so manual artifact component identification is typically required.

In practice we have found (in EEGLAB) that the combination of scalp topography and individual trial raster plots provides the necessary information to identify artifactual components. Therefore a function like ica.plot_components that plotted rasters alongside the scalp topo maps for each IC would be extremely helpful. 

Alex, prior to implementing this if you could share the snippets of code you used to do the raster plots you showed me I'd appreciate it. Thanks!

jona-sassenhagen, 2016-05-29:
Duplicate of #2747

For (semi-)automatically identifying EOG components, consider `ica.find_bads_eog` and `ica.corrmap`. They are explained in the tutorial: http://martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ica.html

(There is still a bug in the Corrmap part.)

agramfort, 2016-05-29:
@espressofiend it's something like

```
ica.get_sources(epochs).plot_image([1, 2, 3])
```

to get the raster for the ICA sources for your epoch.

agramfort, 2016-05-30:
I thought about this. In terms of API it could be:

ica.plot_sources_prop(epochs) -> shows topo, raster and spectrum
ica.plot_sources_prop(raw) -> shows topo and spectrum

wdyt?

jona-sassenhagen, 2016-05-30:
Why not keep the auto segmentation of continuous data EEGLAB does? It's a few lines to do it manually, and really handy IMO.

Also, IMO it should be available for channels too, not just components. If you think of it like that, it makes more sense to use

```ica.get_sources(inst).plot_propertiers([0])```
```inst.plot_propertiers([0])```

espressofiend, 2016-05-30:
From my perspective, way EEGLAB works is useful, however the process of clicking through 20-30 components to see the rasters is a waste of time. A function that plotted the rasters of all components the way `ica.plot_components` does for the topos would be idea IMO - or one that plotted the topos and rasters side-by-side, but required only one call to plot all the components at once. In my experience, the spectra are not informative in making keep/remove decisions for ICA components. 

The ideal solution might be two functions: one that replicates EEGLAB's functionality by showing topo + raster + spectrum for a specified component (or channel, in which case the topo just highlights the selected channel as @jona-sassenhagen suggests in #2747), and one that does what I suggest above, plotting either the rasters of all components/channels in a grid, or plots the topos and rasters side-by-side in a grid. 

@jona-sassenhagen, I've never tried applying `pop_prop` to continuous data but that functionality also sounds very useful.

agramfort, 2016-05-31:
do we have a volunteer to give this a try?

mmagnuski, 2016-05-31:
I'd happily give it a try this week. 
I was thinking about adding some interactivity - keybord left/right and/or left/right buttons to navigate between component properties. So for example `ica.get_sources(inst).plot_properties()` would allow to interactively explore properites of all the components. Or `ica.get_sources(inst).plot_properties(interactive=True)`? WDYT?

agramfort, 2016-05-31:
start by making it work without interaction for now

thanks for volunteering

espressofiend, 2016-05-31:
Yes that's great, thanks! I'm happy to alpha-test if you want.

Aaron

mmagnuski, 2016-06-01:
Ok, I'll start without interaction. 
Besides topoplot, trial-image, erp and spectrum I also plot scatterplot of within-epoch variances (sometimes useful for identifying muscle artifacts (see Hipp 2013) or epochs with strongest component activity (bad epochs / blink epochs etc.)). But if you don't like the variance scatterplot, I can remove it :) 

Here are some examples:
[image]
[image]
[image]
[image]

Let me know if you think the axes layout should be different etc.

espressofiend, 2016-06-01:
I think that looks great! The variance plots are nice. Also the spectrograms are nicer than what's in EEGLAB :)

jona-sassenhagen, 2016-06-01:
Wow, these look great!

agramfort, 2016-06-01:
@mmagnuski can you open a PR? thx

mmagnuski, 2016-06-01:
@agramfort Yes, I'll open the PR today. The implementation is still quite hackish and I'll have at least a couple of questions on API and some details.

mmagnuski, 2016-06-29:
`plot_ica_properties` has been merged, you can plot ic properties this way:
```python
ica.plot_properties(picks=[2, 3, 5])
```
so this issue can be closed.

closed 2016-06-29 by Eric89GXL