mne-tools/mne-python #3274: Q: get mne C / freesurfer to work from python

kingjr, 2016-06-01:
It's probably a bad setup on my part, but I can't seem to make `mne.bem.make_watershed_bem` to work.

I try to explicitly export and source Freesurfer and MNE (C) in the bash and/or via `subprocess` or `os.sys()`, but I keep having the following error with the command

`mne.bem.make_watershed_bem(subject, subjects_dir, overwrite=True)`

```
Running mri_watershed for BEM segmentation with the following parameters:

SUBJECTS_DIR = /data/subjects
SUBJECT = hn120493
Results dir = /data/subjects/hn120493/bem/watershed

Running subprocess: mri_watershed -useSRAS -surf /data/subjects/hn120493/bem/watershed/hn120493 /data/subjects/hn120493/mri/T1.mgz /data/subjects/hn120493/bem/watershed/ws
Command not found: mri_watershed
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-5-f8d51723d992> in <module>()
----> 1 mne.bem.make_watershed_bem(subject, subjects_dir, overwrite=True)

/home/jrking/mne-python/mne/bem.pyc in make_watershed_bem(subject, subjects_dir, overwrite, volume, atlas, gcaatlas, preflood, show, verbose)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    627         with use_log_level(verbose_level):
    628             return function(*args, **kwargs)
--> 629     return function(*args, **kwargs)
    630 
    631 

/home/jrking/mne-python/mne/bem.pyc in make_watershed_bem(subject, subjects_dir, overwrite, volume, atlas, gcaatlas, preflood, show, verbose)
   1061                 'Results dir = %s\n' % (subjects_dir, subject, ws_dir))
   1062     os.makedirs(op.join(ws_dir, 'ws'))
-> 1063     run_subprocess(cmd, env=env, stdout=sys.stdout, stderr=sys.stderr)
   1064 
   1065     if op.isfile(T1_mgz):

/home/jrking/mne-python/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    627         with use_log_level(verbose_level):
    628             return function(*args, **kwargs)
--> 629     return function(*args, **kwargs)
    630 
    631 

/home/jrking/mne-python/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)
    930     logger.info("Running subprocess: %s" % ' '.join(command))
    931     try:
--> 932         p = subprocess.Popen(command, *args, **kwargs)
    933     except Exception:
    934         logger.error('Command not found: %s' % (command[0],))

/home/jrking/anaconda/lib/python2.7/subprocess.pyc in __init__(self, args, bufsize, executable, stdin, stdout, stderr, preexec_fn, close_fds, shell, cwd, env, universal_newlines, startupinfo, creationflags)
    708                                 p2cread, p2cwrite,
    709                                 c2pread, c2pwrite,
--> 710                                 errread, errwrite)
    711         except Exception:
    712             # Preserve original exception in case os.close raises.

/home/jrking/anaconda/lib/python2.7/subprocess.pyc in _execute_child(self, args, executable, preexec_fn, close_fds, cwd, env, universal_newlines, startupinfo, creationflags, shell, to_close, p2cread, p2cwrite, c2pread, c2pwrite, errread, errwrite)
   1333                         raise
   1334                 child_exception = pickle.loads(data)
-> 1335                 raise child_exception
   1336 
   1337 

OSError: [Errno 2] No such file or directory
```

However, when running the command directly in the terminal it works fine
`mri_watershed -useSRAS -surf /data/subjects/hn120493/bem/watershed/hn120493 /data/subjects/hn120493/mri/T1.mgz /data/subjects/hn120493/bem/watershed/ws
Command not found: mri_watershed`.

I can't get the pipeline to work though, because I am still missing the `subject_surface` file.

Any idea what I can do?... Sorry for the noise.

Eric89GXL, 2016-06-01:
> I try to explicitly export and source Freesurfer and MNE (C) in the bash and/or via subprocess or os.sys(), but I keep having the following error with the command

If you do the setup in a `bash` shell, and then start `python` from there, it should work. It sounds like you've tried that, though. You should be able to go through some sequence like:
```Bash
$ which mne_process_raw
$ python -c "import subprocess; subprocess.Popen(['mne_process_raw', '--version'])"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1340, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
$ export MNE_ROOT=~/custombuilds/mne/current
$ source $MNE_ROOT/bin/mne_setup_sh

MNE software location set to:    ...
$ which mne_process_raw
/home/larsoner/custombuilds/mne/current/bin/mne_process_raw
$ python -c "import subprocess; subprocess.Popen(['mne_process_raw', '--version'])"
$ mne_process_raw version 2.35 compiled at Jan 28 2015 06:04:44
```
Can you try it?

> Any idea what I can do?... Sorry for the noise.

My crystal ball suggests you will repay your debt for the noise with a PR to the docs about debugging steps :)

Eric89GXL, 2016-06-01:
(you can adjust the above commands to use `mri_watershed --version` instead, too)

kingjr, 2016-06-01:
`which mne_process_raw`
`/home/jrking/MNE-2.7.4-3452-Linux-x86_64//bin/mne_process_raw`

`python -c "import subprocess; subprocess.Popen(['mne_process_raw', '--version'])"`
`mne_process_raw version 2.36 compiled at Aug  9 2015 04:23:55`

still same error

Eric89GXL, 2016-06-01:
Try it with `mri_watershed --version`

kingjr, 2016-06-01:
Thanks @Eric89GXL . I had (forgottent that I had) saved an (incorrect) environment variable `FREESURFER_HOME` in `~/.bashrc` which is resourced by python `subprocess`.

Not sure how to improve the documentation... I think we should mainly make the error more explicit and make an actual check that freesurfer functions are accessible.

I'll close this one in favor of https://github.com/mne-tools/mne-python/issues/2814

Eric89GXL, 2016-06-01:
"I can't get this MNE-C or FreeSurfer command to run from Python, what do I do?" followed by debugging steps

It's general enough a problem that I don't think will be fully solved by #2814 -- I think we should provide people with these sorts of debugging steps. You could start a PR to document it just by way of this example. In other words, show the error you got, how you debugged it, and what you figured out (you had a path set incorrectly).

kingjr, 2016-06-01:
I think this is may be the origin of my problem 
`set_config` doesn't seem to update `os.environ`:

```
import os
from mne.utils import get_config, set_config
my_path = '/data/subjects'

print('env', os.environ['SUBJECTS_DIR'])
# /usr/local/freesurfer/subjects

print('mne_config', get_config('SUBJECTS_DIR'))
# /usr/local/freesurfer/subjects

print('set to new path', my_path)
# /data/subjects

print('is my new path a dir', os.path.isdir(my_path))
# True

set_config('SUBJECTS_DIR', my_path)

print('mne_config not updated?', get_config('SUBJECTS_DIR')) 
#  '/usr/local/freesurfer/subjects'

print('path not updated?', os.environ['SUBJECTS_DIR'])
# '/usr/local/freesurfer/subjects'

os.environ['SUBJECTS_DIR'] = my_path
print('config manually updated', get_config('SUBJECTS_DIR'))
# '/data/subjects'
```

Eric89GXL, 2016-06-01:
That's by design, it only updates the mne python configuration. I guess any
function we launch we could pass a populated environ dict. We shouldn't
modify os.environ, though.

kingjr, 2016-06-01:
but `set_config` should change the variables in `get_config` shouldn't it?

Eric89GXL, 2016-06-02:
Yeah it should. But that's separate from what should happen to os.environ
(what you mentioned originally)

Eric89GXL, 2016-07-19:
Okay, I finally figured this one out. The precedence for config values is:

1. Options passed into function (e.g., `subjects_dir`)
2. Env vars
3. Config file values

Currently writing config vals with `set_config` does not update the environment, *even if the var is present in the environment*, which gives rise to your problem (i.e., that doing `set_config(...)` seems to have no effect, because it only writes to the config but the var is also present in the env).

I can see a couple of solutions:

1. Add an `update_env=False` default to `set_config`. This is the safest option as it's backward compatible, but I agree can lead to counter-intuitive behaviors.
2. Make `set_config` both write the config file and update the environment. This could be done always, or optionally via `update_env=True` arg.
3. Have update_env=None` -> deprecation currently False to True in 0.14.

I lean toward the third solution. @kingjr that would solve your problems, right? @agramfort is that okay with you?

kingjr, 2016-07-19:
I vote for 3

jona-sassenhagen, 2016-07-19:
FWIW I'm running into the same problem right now.

Eric89GXL, 2016-07-19:
Feel free to try #3439

agramfort, 2016-07-19:
3) is fine with me

PR #3439 (2016-07-20): MRG: Set env vars in set_config
Closes #3274.

Changes `set_config(..., set_env=True)` in 0.14, so that setting config updates `os.environ`, too, by default.
closed 2016-07-20 by Eric89GXL