mne-tools/mne-python #3280: Matplotlib backend failure(s) with 3D Mayavi/surfer/tvtk visualization

kambysese, 2016-06-03:
After installing mayavi 4.4.4 I think I am getting bit by an older [bug](https://mail.scipy.org/pipermail/ipython-user/2012-February/009478.html) having to do with matplotlib backend(s). The error in ipython is:

> ValueError: API 'QString' has already been set to version 1 

with trace...

> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/api.py in <module>()
> ---> 35 from .editors.api import (ArrayEditor, BooleanEditor, ButtonEditor,
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/editors/__init__.py in <module>()
> ---> 22 from .api import (toolkit, ArrayEditor, BooleanEditor, ButtonEditor,
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/editors/api.py in <module>()
> ---> 10 from .code_editor import CodeEditor
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/editors/code_editor.py in <module>()
> ---> 36 class ToolkitEditorFactory ( EditorFactory ):
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/editors/code_editor.py in ToolkitEditorFactory()
> ---> 48     mark_color = Color( 0xECE9D8 )
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traits/traits.pyc in __call__(self, *args, **metadata)
> --> 487         return self.maker_function( *args, **metadata )
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traits/traits.pyc in Color(*args, **metadata)
> -> 1191     return ColorTrait( *args, **metadata )
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/toolkit_traits.pyc in ColorTrait(*args, **traits)
> ----> 7     return toolkit().color_trait( *args, **traits )
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/toolkit.pyc in toolkit(*toolkits)
> --> 117                 _toolkit = _import_toolkit( toolkit_name )
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/toolkit.pyc in _import_toolkit(name)
> ---> 51     return __import__( name, globals=globals(), level=1 ).toolkit
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/traitsui/qt4/__init__.py in <module>()
> ---> 18 import pyface.qt
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/pyface/qt/__init__.py in <module>()
> ---> 28             prepare_pyqt4()
>      29             import PyQt4
>      30             qt_api = 'pyqt'
> /home/ktavabi/anaconda2/lib/python2.7/site-packages/pyface/qt/__init__.py in prepare_pyqt4()
>      15     # Set PySide compatible APIs.
>      16     import sip
> ---> 17     sip.setapi('QString', 2)
>      18     sip.setapi('QVariant', 2)
> ValueError: API 'QString' has already been set to version 1

I can produce the error with
```
import matplotlib.pyplot as plt
import mne
from mne.datasets import sample
data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis-meg'
stc = mne.read_source_estimate(fname)
stc.plot()
```

I am running on

> Platform:      Linux-3.16.0-71-generic-x86_64-with-debian-jessie-sid
> Python:        2.7.11 |Anaconda 4.0.0 (64-bit)| (default, Dec  6 2015, 18:08:32)  [GCC 4.4.7 20120313 (Red Hat 4.4.7-1)]
> Executable:    /home/ktavabi/anaconda2/bin/python
> 
> mne:           0.13.dev0
> numpy:         1.10.4 {lapack=mkl_lapack95_lp64, blas=mkl_intel_lp64}
> scipy:         0.17.0
> matplotlib:    1.5.1
> 
> sklearn:       0.17.1
> nibabel:       2.0.2
> nitime:        Not found
> mayavi:        4.4.4
> nose:          1.3.7
> pandas:        0.18.0
> pycuda:        2016.1
> skcuda:        0.5.2

cc @kingjr @Eric89GXL

agramfort, 2016-06-04:
I do:

    export ETS_TOOLKIT=qt4
    export QT_API=pyqt

and it works for me.

kingjr, 2016-06-04:
`export ETS_TOOLKIT=qt4` stopped the `core.dump` error when opening more than one window for me.

kambysese, 2016-06-04:
Thanks will try. You putting this in `.bashrc`, or existing shell?

agramfort, 2016-06-04:
I have this in my bashrc

kambysese, 2016-06-06:
No such luck for me :( Same error with above code snippet in ipython console; as well as IDE.

kambysese, 2016-06-27:
@agramfort  I stand corrected. The `env` variables do work.

closed 2016-06-27 by kambysese