mne-tools/mne-python #3281: BUG: write_surface doesn't write in same reference frame as read_surface

kingjr, 2016-06-04:
`write_surface` doesn't seem to write in the correct coordinates system. However, this is visible in freeviewer not in the mne `viz` functions.

To reproduce the error (you have to copy/paste `subjects/sample` into `subjects/sample_test` since it modifies the file in a bad way.)

```
import os.path as op
from mne.viz import plot_bem
from mne.datasets import sample
from mne.surface import read_surface, write_surface
data_path = sample.data_path()

# the raw file containing the channel location + types
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
# The paths to freesurfer reconstructions
subjects_dir = data_path + '/subjects'
subject = 'sample_test'

fname = op.join(subjects_dir, subject, 'bem', 'inner_skull.surf')
coord, faces = read_surface(fname)
write_surface(fname, coord, faces)

plot_bem(subject=subject, subjects_dir=subjects_dir, orientation='coronal')
```

`plot_bem` plots as expected:

[image]

but in freeview, we now see a misalignment:

[image]

Eric89GXL, 2016-06-04:
Can you try using nibabel to read and write the surfaces to see if they
have the same problem?

kingjr, 2016-06-04:
What is the nibabel function to read surfaces?

Eric89GXL, 2016-06-04:
From Googling "nibabel read freesurfer surface":

http://nipy.org/nibabel/reference/nibabel.freesurfer.html#nibabel.freesurfer.io.read_geometry

kingjr, 2016-06-04:
> From Googling "nibabel read freesurfer surface":

ouch.

Yes same error, I'll open an issue there.

Eric89GXL, 2016-06-04:
We might have contributed the code there actually, not sure :) I'll try to
take a look Monday. I've seen something like this before, but assumed it
was a Freeview problem. I'm not aware of any coordinate frame information
on the file, and we just do simple read and write with no transformations,
so I'm not sure why it would happen.

The original (not yet overwritten) surface looks okay though, right?

kingjr, 2016-06-04:
yes they look fine.

closed 2016-06-29 by agramfort