mne-tools/mne-python #3282: BUG: save coregistration with MRI scaling

kingjr, 2016-06-04:
When applying an MRI scaling during the coregistration, we only have the option to save as a different subject name, which creates a `OSError` in the background (not a python crash) when saving because the corresponding directory doesn't exists. I guess this is a partly intended feature, but it still needs a fix.

To reproduce the issue:

```
from mne.gui import coregistration
from mne.datasets import sample
data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
subject = 'sample'
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
coregistration(subject=subject, subjects_dir=subjects_dir, inst=raw_fname)
```
=> apply `mri scaling`
=> save as `sample_foo`
=> `No such file or directory: '/home/jrking/mne_data/MNE-sample-data/subjects/sample_foo/bem'`

kingjr, 2016-06-04:
Also, when you tick `overwrite` you cannot untick it. (This is a separate issue, but since I'm opening many, I'd rather say this here...)

kingjr, 2016-06-04:
(also, the subsequent scaling of the bem solution is incredibly slow for a linear scaling.)

christianbrodbeck, 2016-06-24:
> Also, when you tick overwrite you cannot untick it. (This is a separate issue, but since I'm opening many, I'd rather say this here...)

This one should be fixed now with a hack (https://github.com/mne-tools/mne-python/pull/3332/commits/1d37a05d7d9e003dd6f805035598bc8ade77e421) but I don't get what the problem is. I've tried to simplify the issue in that gist: https://gist.github.com/anonymous/ffa91548872ad8efbaaa046c5f86eb62

When I run it I see:
```
Paul -> Peter
Paul -> Peter
 -> Peter
 -> Peter
 -> Peter
 -> Peter
```
so `subject_from` gets set correctly at the beginning, but then it's suddenly gone. On clicking the overwrite button it comes back:
```
Paul -> Peter
Paul -> Peter
```
and that's why the overwrite button behaves the way it does...

closed 2016-07-13 by christianbrodbeck