mne-tools/mne-python #3283: Q: Issue with source reconstruction on my BEM but not sample's

kingjr, 2016-06-04:
I am struggling to reconstruct the source of a presumably simple dataset. 

The evoked topo (here at 180ms) looks dipolar, and should come from the back of the brain (response to visual stim).

[image]

When I reconstruct the source on with the `sample` BEM, i get something relatively expected  (to the extent that it's in the back of the brain.

[image]

However, when I try to reconstruct on my subject's actual bem, I get very unlikely reconstructions: Here MNE at 180ms reconstruct a frontal activity (similar with dSPM and sLORETA, in mne-python or mne-c):
[image]
[image]

I checked that the source space and `trans` alignment make sense (here dipole fit at 180 ms):

[image]
[image]

I'm using the same evoked and covariance in both cases. 

I encounter this issue with all of my subjects. 

Any idea in what direction I search?

[I can share the data in private, I haven't defaced the MRI]

agramfort, 2016-06-04:
Check the sensitivity maps of your forward

kingjr, 2016-06-06:
The error was coming from the covariance estimation.

I believe this is because I am computing the covariance across all epochs, although they are concatenated from different runs, (and therefore were maxfiltered separately, but supposedly head realigned).

Here is the covariance ranks computed on 168 epochs from a single run:
[image]

and the sources found:
[image]

Here is the covariance ranks computed from 168 epochs from one run and 2 trials from another run.
[image]

and the corresponding sources
[image]

The issue is independent of the covariance's method.

Thanks hips to @Eric89GXL for his help! (Turned out that it work with your data because I had cropped it to make it lighter)

I think we should add a warning on the `compute_covariance` when we detect that the epochs are not from the same run (if it's even possible?).

Eric89GXL, 2016-06-06:
We should at least emit a warning, yeah. The `info['dev_head_t']` at the very least should be checked for consistency, that presumably would have caught your problem.

Presumably using the Maxwell filter head translation to the same coordinate frame would also solve this problem.

kingjr, 2016-06-06:
> Presumably using the Maxwell filter head translation to the same coordinate frame would also solve this problem.

I thought that was already applied on my data. I don't have access to the raw and didn't do this step myself. Is there a way to check it out and reprocess it?

Eric89GXL, 2016-06-06:
Reprocessing would probably be a pain, but you could try it. You can check by seeing if `info['dev_head_t']` match across all files.

kingjr, 2016-06-06:
>  You can check by seeing if info['dev_head_t'] match across all files.

They don't :( 

> Reprocessing would probably be a pain, but you could try it.

By 'pain' you mean that it would be long, or would it cause processing issues? Would it be fine to do this with MNE on sssed data?

agramfort, 2016-06-06:
calling evoked.plot_white would have helped too :)

kingjr, 2016-06-06:
@agramfort mmmh, I don't think so.

This is the whitened evoked with the cov computed from a single run
[image]

This is the whitened evoked with the cov computed from multiple runs
[image]

agramfort, 2016-06-06:
hum overfit the baseline and your noise cov is not regularized at all...
can you try with the shrunk method? for me own personal interest and
certainly @dengemann :)

Eric89GXL, 2016-06-06:
I suspect it won't help too much because the eigenvalue spectrum's falloff is no longer that steep due to the different head positions. But maybe it's robust enough to do it.

agramfort, 2016-06-06:
worth trying but I agree not the proper fix. I am just curious

kingjr, 2016-06-06:
For this subject, `shrunk ` works nicely if I give all the trials (n=900); however it seems that if I only give a few trials from an extra block, I retrieve weird frontal activity again; So I guess it is indeed SNR issue.

I had originally used the shrunk method in my pipeline, and several subjects had crazy activations as described here, this is why I started to dig the issue.

Also, FYI the whitened evoked with shrunk with 900 trials  gets values outside the plot limit (we actually don't see anything in the default plot, including in the baseline).

kingjr, 2016-06-06:
Any chance I can realign the heads directly on epochs data? Just to check whether it's worth it...

dgwakeman, 2016-06-06:
You could source analyze the runs separately, and then average the source reconstructed data. But with as much motion as there must be to have this much of an issue, I would trash the data. There are some data collection issues, that just arent recoverable IMO

kingjr, 2016-06-06:
@dgwakeman ok this seems indeed suboptimal. If I have time I'll try to get the raw and realign from the beginning.