mne-tools/mne-python #3284: correct the API usage of mask in mne.find_events to be consistent with mne-c, and more user-friendly

teonbrooks, 2016-06-05:
while I was addressing a comment on the listserv, I noticed that although the masking is correct per its instructions, it is not consistent with the mne-c way of handling the trigger mask. I also take blame because I added the feature. in essence, I inverted the meaning of the ones and zeros, which makes constructing the mask much more difficult.
 this is how it should be:
> --digtrigmask <number>
> Mask to be applied to the trigger channel values before considering
> them. This option is useful if one wants to set some bits in a don’t
> care state. For example, some finger response pads keep the trigger
> lines high if not in use, i.e., a finger is not in place. Yet, it is convenient
> to keep these devices permanently connected to the acquisition
> system. The number can be given in decimal or hexadecimal format
> (beginning with 0x or 0X). For example, the value 255 (0xFF)
> means that only the lowest order byte (usually trigger lines 1 - 8 or
> bits 0 - 7) will be considered.

I propose that we make it our mask option consistent with this. it makes its use case more user-friendly, and fixes the inconsistency issue. 

a real example, a user's triggers ranged from 0-32 and are only described on the first six bits of the trigger channels. in order to isolate these values, we need to mask the upper bits. the current implementation requires `mask=2**17 -32` whereas the mne-c way, and the way I propose would have it be `mask=32`.

What are your thoughts?
cc: @agramfort, @Eric89GXL

agramfort, 2016-06-05:
fine with me if we don't break people's code without proper deprecation
cycle

Eric89GXL, 2016-06-06:
I think there will need to be a deprecation cycle, but I think it's worth it to get equivalent to MNE-C.

We could introduce a `kwarg` like `kind='and' | 'and not'` (or some better names) for the kind of masking to do. For one cycle it can be `kind=None` which wraps to the existing `'and not'` behavior, changing to `kind='and'` for 0.14.

PR #3500 (2016-08-09): MRG: kwarg mask_type like MNE-C
Closes #3284
closed 2016-08-09 by Eric89GXL