mne-tools/mne-python #3286: ENH: detect whether dev_t is the same across epochs before compute_covariance

kingjr, 2016-06-06:
In https://github.com/mne-tools/mne-python/issues/3283, I encountered a difficulty to find what was wrong with my sources. 

This was because the covariance was estimated from trials acquired in different run, and therefore with different `dev_head_t`. 

`compute_covariance` should probably throw a warning if the head alignment don't match.

cc @dengemann

teonbrooks, 2016-06-07:
related to #2109 and #2122. realigning isn't as straightforward as one would think...

started down the road last year with @yousrabk but needed to retreat.

would be happy to see a solution ;)

kingjr, 2016-06-07:
+1, but I think we should first start with a warning, that should be easy enough ("PR welcome")

Eric89GXL, 2016-06-08:
Actually an error by default would be better. We can have an argument `misaligned='error' | 'warning'` for now and eventually add `misaligned='interpolate'`. Maybe there is a better name than `misaligned`, though.

Eric89GXL, 2016-06-08:
I loosely plan to work on reconstruction (#3184) this summer, so I might be able to tackle the interpolation bit.

PR #3313 (2016-06-22): MRG: Check MEG<->Head transform during cov calculation
@kingjr can you have a look to see if this would have saved you some pain?

Closes #3286.
closed 2016-06-22 by agramfort