mne-tools/mne-python #3287: ENH: evoked.decim

Eric89GXL, 2016-06-07:
It would be nice to be able to decimate evoked data like we can with `Epochs`. It should be fairly straightforward.

kingjr, 2016-06-07:
We could put it in a Mixin so that other classes would benefit from it (e.g. stcs, PSDs).

(BTW, I'm still in favour of adding it to `Raw` too, it really make life easier to jump from laptop/digging to cluster/production mode)

Eric89GXL, 2016-06-07:
> We could put it in a Mixin so that other classes would benefit from it (e.g. stcs, PSDs).

We could, but the logic is actually quite different across classes. I'm implementing it now for `Evoked` and the shared logic is mostly in checking the parameters, so I've refactored that into a function. The docstring is pretty similar, though.

> I'm still in favour of adding it to Raw too

This would be pretty easy with `preload=True` (which you'd probably want anyway, since you should probably have filtered or resampled to use `decimate`) but kind of a pain with `preload=False`.

agramfort, 2016-06-07:
no objection

kingjr, 2016-06-07:
Yes for raw, maybe later.

> We could, but the logic is actually quite different across classes. 

For Epochs, Evoked, STCs & PSDs you can always get `info`, `times` and do `data[..., ::decim]` no?

Eric89GXL, 2016-06-08:
Sometimes it's .data, sometimes ._data; sometimes it's preloaded sometimes
not; sometimes 3D, sometimes 2D; for TFR the aliasing warning might need to
be modified; for STC we have no low pass information at all; there are
inconsistencies with how to set times, etc. And if you add Raw to the mix
it gets worse...

None of these problems are insurmountable, but I'm not sure it's worth
tackling them for now.

kingjr, 2016-06-08:
You're right. This probably points out to a more general issue of class consistencies. Forget my proposal.

PR #3289 (2016-06-10): MRG: Add Evoked.decimate
Also fixes a bug with existing `Epochs.decimate` when the `offset` parameter is used.

Closes #3287.
Closes #3296.

Ready for review/merge from my end.
closed 2016-06-10 by Eric89GXL