mne-tools/mne-python #3291: ENH: Bem editing?

Eric89GXL, 2016-06-08:
@kingjr recently asked (but is probably too wary of opening issues to do it):

> is there a way to quickly edit BEMs? I have some intersections between inner and outer skulls but they're tiny, and a manual dragging of the points would really help.

My suggestion would be to:

1. Find some nice triangulated mesh editing software (3D program of some sort, maybe CAD or blender?)
2. Figure out what file format it needs
3. Convert the surf to that format (it should be simple, if it's not, open MNE-Python issue and I'll help there)
4. Document the process in the docs

Does anyone know of a better way? If not, then we can make a recommendation for how people can fix minor BEM surface issues if necessary.

kingjr, 2016-06-08:
Relatedly, I am currently trying to see whether I can fix the tiny surface intersections problem by:
* Detecting the points outside the outer surface (e.g. inner skull point out of outer skull)
* Changing the xyz of these points by a param `k` toward the center of the surface

dgwakeman, 2016-06-08:
1. Make sure people know they should be using good quality MR sequences including MEFLASH if they have EEG
2. If they just have MEG they should just use a single layer
3. in lucky cases mne_surf2bem can shift the surfaces to avoid simple problems

Eric89GXL, 2016-06-08:
@kingjr you know we have code for detecting whether points are inside a surface already right?

kingjr, 2016-06-08:
@Eric89GXL Yes, I'm now trying to understand how it works :)

Eric89GXL, 2016-06-08:
Follow the trail of the [surface topology checks in _check_surfaces] (https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L427)

dgwakeman, 2016-06-08:
Seriously though, is there a problem with getting an MEFLASH or with mri_make_bem_surfaces (it has never failed for me)?

kingjr, 2016-06-08:
@dgwakeman  It's not always possible to acquire MRI, sometimes you have to retrieve from archive data.

kingjr, 2016-06-08:
[i.e. you don't always have the adequate MRI sequence, hence the unstable segmentation]

dgwakeman, 2016-06-08:
Ok, in that case the one cheap easy step to solve the intersection problem with the watershed BEM is to just binarize and smooth/expand the 3d volume of the aseg volume from FreeSurfer, and then just fit an icosahedral surface to it. That should avoid all intersections. This would be totally automatable, and easy. I had some hacks to do this back in the day (hacking calls to mri_watershed: the code is pretty entrenched in a particular naming directory structure). Note, the watershed is still not recommended for EEG.

kingjr, 2016-06-08:
@dgwakeman that seems like a better fix than I was suggesting. Would you be ok sharing your script and opening a PR?

dgwakeman, 2016-06-08:
I can try to dig it up, but it doesn't really belong in mne-python (unless we got icosahedral meshing and I missed it), because it uses FreeSurfer commands. I'm happy to help. I'm in the middle of moving (to Philadelphia if anyone knows of labs looking for someone, or wants to hire a consultant to work remotely), but I should be able to look into it late next week. I'm setting an alarm now to look at it on Friday afternoon. hth d

romquentin, 2016-11-23:
If I understand correctly, because I have only MEG data, I can only reconstruct a one layer bem model with
`make_bem_model(subject, conductivity=[0.3], subjects_dir=subjects_dir)` which will give me only the inner_skull surface?

kingjr, 2016-11-23:
yes, for meg you don't need 3 layers

romquentin, 2016-11-23:
Ok Thank you!
On some case, the inner skull miss a part of the brain so I would still need a tool to expand the inner skull surface...

romquentin, 2016-11-23:
Specifically, this pattern is very common in my participants, with a surface error on the right frontal part... 
[image]

larsoner, 2017-12-05:
Closing as I don't think there is much we can do in MNE to fix this, it's more of a Freesurfer / triangulated mesh problem (but feel free to comment further if people find other fixes)

closed 2017-12-05 by larsoner

nwilming, 2018-03-05:
Maybe this is useful for future reference. 

I had a similar problem to @romquentin, one of my outer skin meshes had fairly large holes on either side. This was actually pretty easy to fix using meshlab (http://www.meshlab.net/). This is what I did:
 1. Convert freesurfrer surfaces to e.g. stl format using mris_convert
 2. Open surface in meshlab
 3. In my case I selected the faces in my 'holes' using the selection tool, making sure to press 'alt' to only select visible faces.
 4. Delete offending faces
 5. Select 'Filters -> Remeshing, Simplification and Reconstruction -> Close holes'
 6. Export surface to stl file. Make sure to uncheck the 'Binary encoding' checkbox.
 7. Convert stl surface to freesurfer surface format using mris_convert.

This nicely filled the holes. Not sure how helpful meshlab is for other defects but is certainly worth a look.

larsoner, 2018-03-05:
@nwilming nice! It would be good to add this to the docs. We could either put it in the FAQ:

https://martinos.org/mne/stable/faq.html

Or here:

https://martinos.org/mne/stable/manual/source_localization/forward.html#topology-checks

The latter is a bit outdated / talks about MNE-C a lot. We should probably update it. But that wouldn't prevent us from adding this useful information there.

ezemikulan, 2018-05-11:
Hi, I've been trying different ways to edit the surfaces used for BEM. In my experience, the *.stl* format is not useful for further processing as it loses the vertex order information, which results in an error when creating the BEM model (*mne.make_bem_model*):

*"RuntimeError: The source surface has a matching number of triangles but ordering is wrong".* bem.py (line 361)

The way I found to edit the surfaces is:

1. Convert from freesurfer surface to ascii (with freesurfer's mris_convert)
2. Convert from ascii to .obj (using [Brainder's](https://brainder.org/2012/05/08/importing-freesurfer-cortical-meshes-into-blender/)  functions)
3. Edit in Blender (and export .obj)
4. Convert from .obj to ascii (using [Brainder's](https://brainder.org/2012/05/08/importing-freesurfer-cortical-meshes-into-blender/)  functions)
5. Convert from ascii to freesurfer surface (with freesurfer's mris_convert)

I createad a small [guide](https://github.com/ezemikulan/blender_freesurfer) that shows how to do the edition in Blender and also some scripts to automatize the process of converting all the files created by *mne.bem.make_watershed_bem*.

I hope it will be useful!