mne-tools/mne-python #3292: ENH: save morph sparse matrix

kingjr, 2016-06-08:
The MNE C script [mne_make_morph_maps](http://mne-tools.github.io/stable/manual/c_reference.html#mne-make-morph-maps) generates `fif`files, but I haven't found documentation for the MNE Python [example](http://mne-tools.github.io/stable/auto_examples/inverse/plot_morph_data.html) or [FAQ](http://mne-tools.github.io/stable/faq.html?highlight=morph#morphing-data).

However, saving sparse compressed raw matrices isn't [trivial](http://stackoverflow.com/questions/8955448/save-load-scipy-sparse-csr-matrix-in-portable-data-format), as saving and load with numpy doesn't crash, but also doesn't save properly (it saves as `np.array(dtype=object)` instead).

I'm not sure whether we want to handle this in [h5io](https://github.com/h5io/h5io/issues/16), make a `Morph` class to add a `save` routine or something else.

Eric89GXL, 2016-06-08:
We have code to do it in Python but it's slower. Also, we can read the morph maps directly as they are FIF files, which can already store sparse matrices.

kingjr, 2016-06-08:
So once you've done:

```
morph = mne.compute_morph_matrix(subject_from, subject_to, vertices_from, vertices_to)
```

What do you do to save `morph` as a fif?

Eric89GXL, 2016-06-08:
You shouldn't manually compute it, use http://mne-tools.github.io/stable/generated/mne.read_morph_map.html#mne.read_morph_map

Eric89GXL, 2016-06-08:
Oh wait, sorry, you're talking about a morphing matrix which is different from a morph map. We don't currently have a way of saving the morphing matrix, no.

kingjr, 2016-06-08:
Yes, sorry for the bad terminology

kingjr, 2016-06-08:
So +1 on writing `mne.source_estimate.save_morph_matrix()` and `mne.source_estimate.load_morph_matrix()`? Shall we use `.fif` or h5io?

Eric89GXL, 2016-06-08:
It might actually be nice to turn the morph matrix into a proper object, actually. Store the `subject_from` and `subject_to`, as well as `vertices_from` and `vertices_to`. It could save some user mistakes I think. Then it could just be `stc_fsaverage = stc_subj.morph_precomputed(morpher)`, where `morph_precomputed` can deal with getting a `MorphMatrix` object, and does proper sanity checks.

Eric89GXL, 2016-06-08:
...in which case I think we'd probably want to use HDF5 because I don't think we'd have the correct tags in FIF land.

agramfort, 2016-06-09:
no objection with h5io

nfoti, 2016-06-27:
A similar discussion is what prompted PR #3334. I'm running into a similar situation where I'm mapping a lot of epochs from a number of subjects to `fsaverage` and the only reasonably time/space-efficient way to store the resulting source space data is to save the morph matrices and to morph the epochs on demand. Having an object to store the matrices and the `vertices_from` and `vertices_to` would be incredibly helpful.

Eric89GXL, 2016-06-28:
I propose adding a new class like `Morpher` or `MorphMatrix` for this purpose. It would store the vertices_from, vertices_to, and matrix, and do input validation and the like when it's used. @agramfort sound good to you? If so I can take a stab at it.

agramfort, 2016-06-29:
can you write me a gist of how you image the code will look like?

larsoner, 2019-03-27:
We have this now

closed 2019-03-27 by larsoner