mne-tools/mne-python #3294: DOC: Use class names in docstring

Eric89GXL, 2016-06-08:
I noticed that we have this sort of thing in a lot of places:
```
Returns
-------
info : instance of Info
```
If we replace it with this, the class name becomes a nice hyperlink:
```
info : instance of :class:`Info`
```
I think we should consider doing this from now on.

christianbrodbeck, 2016-06-27:
It makes the raw doctoring somewhat less readable, but the links would be nice – Since numpydoc standard is `name : type` I wonder whether the link could be done in sphinx somehow even without the `:class:` notation?

Eric89GXL, 2016-06-27:
That would be a potential enhancement to numpydoc repo. Not sure if
"instance of X" is ubiquitous enough to get that type of thing merged
though. If it is, it might not be so hard to do.

Eric89GXL, 2016-06-28:
See:

https://github.com/numpy/numpydoc/issues/43

agramfort, 2016-06-28:
it would be great to autodetect classes