mne-tools/mne-python #3296: BUG: problems with epochs._raw_times while cropping resampled epochs

joewalter, 2016-06-09:
While cropping epochs, we also change epochs._raw_times. However, as soon as epochs were resampled before cropping, epochs._raw_times is based on a different sfreq. Thats why we cannot apply the same tmask for epochs.times and epochs._raw_times. This leeds to wrong values in epochs._raw_times if sfreq_new < sfreq_old and an error if sfreq_new > sfreq_old. Changing epochs.crop(). We thus should use a different tmask for changing epochs._raw_times. As we do not store the old sampling frequency (AFAIK), we need to compute it as 1./ (epochs._raw_times[1]-epochs._raw_times[0]).

In mne.epochs.crop(), changing
```
tmask = _time_mask(self.times, tmin, tmax, sfreq=self.info['sfreq'])
self.times = self.times[tmask]
self._raw_times = self._raw_times[tmask]
self._data = self._data[:, :, tmask]
```
to 
```
tmask = _time_mask(self.times, tmin, tmax, sfreq=self.info['sfreq'])
self.times = self.times[tmask]
self._data = self._data[:, :, tmask]
tmask = _time_mask(self._raw_times, tmin, tmax,
    sfreq=(1. / (self._raw_times[1] - self._raw_times[0])))
self._raw_times = self._raw_times[tmask]
```
should solve the issue.

agramfort, 2016-06-09:
thx @joewalter for the bug report.

@jaeilepp can you have look?

jaeilepp, 2016-06-09:
Another option is to update ``epochs._raw_times`` when resampling. I'm not sure why we have this ``_raw_times`` attribute. It seems redundant with the times attribute.

agramfort, 2016-06-09:
no clear for me either

Eric89GXL, 2016-06-09:
It might be necessary, see the decimate code and my recent PR and you'll
see how it's used under the hood.

After loading data, ._raw_times should always be the same as .times (it's
only used to know what to read from disk), so an easier fix is to update it
during resampling as is done in decimation.

Eric89GXL, 2016-06-09:
Also, keep in mind decimation is preferred to resampling where possible
(i.e., when raw data was appropriately lowpassed) to avoid potentially ugly
edge artifacts.

Eric89GXL, 2016-06-09:
I can add a fix for this in my Evoked decim PR unless you're already
working on it @jaeilepp

jaeilepp, 2016-06-09:
It's a one liner, so go ahead.

PR #3289 (2016-06-10): MRG: Add Evoked.decimate
Also fixes a bug with existing `Epochs.decimate` when the `offset` parameter is used.

Closes #3287.
Closes #3296.

Ready for review/merge from my end.
closed 2016-06-10 by Eric89GXL