mne-tools/mne-python #3299: 'predict_proba' is not an option in GeneralizationAcrossTime

romquentin, 2016-06-10:
@kingjr 
When running this:
`gat = GeneralizationAcrossTime(clf=clf, scorer=scorer, n_jobs=-1, predict_mode='predict_proba')`
`gat.fit(epochs, y)`
`gat.score(epochs, y=y)`

It gives this error:
`ValueError: `predict_mode` must be a str, "mean-prediction" or"cross-validation"`

I have the last mne update. Did you remove the option?

Thanks

jona-sassenhagen, 2016-06-10:
You're confusing predict_mode and predict_method. Check the docs again: http://martinos.org/mne/stable/generated/mne.decoding.GeneralizationAcrossTime.html#mne.decoding.GeneralizationAcrossTime

Admittedly, this is confusing nomenclature.

romquentin, 2016-06-10:
Thanks. But now running this:
`gat = GeneralizationAcrossTime(clf=clf, scorer=scorer, n_jobs=-1, predict_method='predict_proba')`
gives this error:
`TypeError: __init__() got an unexpected keyword argument 'predict_method'`
What I am doing wrong?

kingjr, 2016-06-10:
Which version of mne are you using?

kingjr, 2016-06-10:
> Admittedly, this is confusing nomenclature.

Yes. I'll refactor the gat in the late summer. sklearn will soon accept `predict_proba` for its pipeline so we won't need this anymore.

romquentin, 2016-06-10:
I cloned the last version but I guess something was wrong with the install. I reinstall with `pip install mne --upgrade`and it works. 
Thanks!

closed 2016-06-10 by romquentin