mne-tools/mne-python #3300: brain.contour_list not updated in split view

kingjr, 2016-06-10:
When plotting contours with a `split` view and both hemisphere, plotting a contour on each hemisphere replaces `brain.contour_list` instead of appending the contours; the contour handle thus gets lost:

e.g.

```
import numpy as np
import mne

data_path = mne.datasets.sample.data_path()
subjects_dir = data_path + '/subjects'
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
fname_event = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

# Load data
inv = mne.minimum_norm.read_inverse_operator(fname_inv)
raw = mne.io.read_raw_fif(fname_raw)
events = mne.read_events(fname_event)

# Read epochs
epochs = mne.Epochs(raw, events, 4, -.2, .3)
evoked = epochs.average()

stc = mne.minimum_norm.apply_inverse(evoked, inv, .1, method='dSPM')
brain = stc.plot(subject='fsaverage', hemi='split', views=['lat', 'med'])

brain.add_contour_overlay(np.random.rand(163842), hemi='lh')
brain.add_contour_overlay(np.random.rand(163842), hemi='rh')
for contour in brain.contour_list:
    contour['surface'].remove()
# only the contours of the right hemisphere get removed.
```

Eric89GXL, 2016-06-10:
Is this a problem with `stc.plot` or `brain.*`? If it's the latter, it's a PySurfer issue, not MNE-Python

kingjr, 2016-06-10:
I am not sure because it seems specific to `hemi='split'`.

Eric89GXL, 2016-06-10:
`hemi='split'` is a mostly PySurfer option under the hood

closed 2016-06-10 by Eric89GXL