mne-tools/mne-python #3301: circular_source_labels?

Eric89GXL, 2016-06-11:
There is an instance in our code examples in `mne/simulation/source.py` of `circular_source_labels`. I'm not sure where it comes from, since we don't have such a function. If anyone knows please say so. Otherwise I'll do some `git blame` work to track it down when I find time.

Eric89GXL, 2016-06-11:
cc @nfoti found this one

agramfort, 2016-06-11:
@nfoti are you on it?

nfoti, 2016-06-11:
I'm happy to try to add it.

Eric89GXL, 2016-06-11:
Apparently back in 2012 (!) we actually had such a function, which I found using `blame` on GitHub:

https://github.com/mne-tools/mne-python/commit/5e66a3890b851ca4ab6184d830579cb817ca8869

It looks suspiciously like `grow_labels`, so I suspect it's just a renaming that we missed:

https://github.com/mne-tools/mne-python/blob/5e66a3890b851ca4ab6184d830579cb817ca8869/mne/simulation/source.py#L259

Eric89GXL, 2016-06-11:
Found it -- back in 2012 someone not-to-be named (hint: last name rhymes with shamdort) refactored `circular_source_labels` to `grow_labels` but must have missed a reference in the docstring, probably because the docstring wasn't written properly in the first place to execute as a `doctest`:

https://github.com/mne-tools/mne-python/commit/4c12dd9b60ec08690cb7b98a1bb15bc9a97b27c0

agramfort, 2016-06-12:
>
> Found it -- back in 2012 someone not-to-be named (hint: last name rhymes
> with shamdort) refactored circular_source_labels to grow_labels but must
> have missed a reference in the docstring
>
naughty  shamdort....

nfoti, 2016-06-20:
It doesn't look like the docstring of `grow_labels` needs changing, I think you mean the docstring of `simulate_stc`. I'm not certain that `grow_labels` can be used though since it doesn't seem to actually fill in the `value` field of the generated `Label` (I could be mistaken about this though). I could add in an example that uses the new `Label.center_of_mass` function that will select the vertex closest to the center of mass. Basically the following, given a `Label` called `label`:
```
com = label.center_of_mass('sample', surf='white')
label.values = np.zeros_like(label.values)
label.values[np.where(label.vertices == com)[0][0]] = 1
```

Eric89GXL, 2016-06-20:
>  I think you mean the docstring of simulate_stc

Yeah wherever had the `circular_source_labels` reference.

>  I'm not certain that grow_labels can be used though since it doesn't seem to actually fill in the value field of the generated Label (I could be mistaken about this though).

Ahh right, because you need it to be the distance away from the original seed? I think for filling in the values afterward you'll need to use `dijkstra` from SciPy manually. I can't think of a clean API to offer for the users that would see enough use to include it. It would be nice in an example, though.

nfoti, 2016-06-21:
Agreed. Do we just want to remove the reference to `circular_source_labels` from the `simulate_stc` docstring and close this?

The point spread example I'm writing for #3308 can demonstrate how to use the values in labels in `simulate_stc` as well as the new `center` mode for `simulate_sparse_stc`.

Eric89GXL, 2016-06-22:
Yeah, please fix the docstring as part of your example, I forgot to do it.

commit f5e56a95 (2016-09-08): FIX: Fix docstring (closes #3301)
closed 2016-09-08 by Eric89GXL