mne-tools/mne-python #3302: BUG: Error when brainvision data contains channels with units different to V or uV

jscastanoc, 2016-06-11:


agramfort, 2016-06-11:
what version are you using?

can you test with current master? bug about this was fixed recently

jscastanoc, 2016-06-11:
Hi Alexandre,

I am using 0.13.dev0, I cloned it yesterday.

I have a temporal fix:
mne/io/brainvision/brainvision.py line 353:

change
ranges[n] = _unit_dict.get(unit, unit) * scale
for:
ranges[n] = _unit_dict.get(unit, 1) * scale

In our setup we have an extra channel with units "N" (Newtons), I am unsure if I would break something with my temporal fix. Should I try a pull request?

agramfort, 2016-06-12:
@jscastanoc can you share one file with me and @jaeilepp 

@jaeilepp can certainly do the fix on monday.

jscastanoc, 2016-06-12:
@agramfort just send you a piece of the data. Could you please forward it to @jaeilepp ?

Thanks!

closed 2016-06-13 by agramfort