mne-tools/mne-python #3305: SNR in calculation in `add_noise_evoked` might be incorrect

clamus, 2016-06-13:
I'm trying to see how the data is scaled to achieve a given SNR in `add_noise_evoked` in  [mne-python/mne/simulation/evoked.py](mne-python/mne/simulation/evoked.py), but do not really get it.

The SNR in this function is defined as the standard SNR in dB, which is:
```math
SNR = 10 * log10 [ var(signal) / var(noise) ]
```

The piece of code that should do this is:

```python
tmp = 10 * np.log10(np.mean((evoked.data[:, tmask] ** 2).ravel()) / np.mean((noise.data ** 2).ravel()))
noise.data = 10 ** ((tmp - float(snr)) / 20) * noise.data
```

However, I am think that it should be instead:

```python
tmp =  np.log10(np.mean((evoked.data[:, tmask] ** 2).ravel()) / np.mean((noise.data ** 2).ravel()))
noise.data = 10 ** (tmp - float(snr) / 10) * noise.data
```

Is this a bug, or am I missing something?

kingjr, 2016-06-13:
The line was apparently modified in https://github.com/mne-tools/mne-python/pull/68: https://github.com/mne-tools/mne-python/commit/0a4c44ba414a65da877e4e36ba53022b0cbd9bb3,  from https://github.com/mne-tools/mne-python/commit/c339669fe324e0d4d7262abed68afd81affa3c8c

Maybe @agramfort @mluessi or @joewalter can clarify this

Eric89GXL, 2016-06-13:
@clamus to simplify your proposal you could equivalently change a single line (you had changed 2) as:
```
noise.data = 10 ** ((tmp - float(snr)) / 10) * noise.data
```
Is that right?

clamus, 2016-06-13:
@Eric89GXL that is correct. Changing just that line is fixes the issue in a better way. I think the issue came from where @kingjr was pointing to.

clamus, 2016-06-13:
Should I go ahead and fix this?

agramfort, 2016-06-13:
go for it ! thx

clamus, 2016-06-15:
Oops... It turn out the original code was correct :$ One does need to divide by 20, which scales correctly with the square of the noise. It was a good thing that @agramfort suggested I added a test, which showed I made a mistake. 

@agramfort Should I still add the test? What about the what's new section?

agramfort, 2016-06-15:
lol :)

yes please just add a test but don't update what's new

thx

closed 2016-06-16 by clamus