mne-tools/mne-python #3307: Modes for `simulate_sparse_stc`

nfoti, 2016-06-14:
Currently the `simulate_sparse_stc` function chooses a random vertex in each label to set to the desired signal. It would be useful to also have a mode to use the vertex in each label that is closest (in Euclidean distance) to the center of mass of the label.

I have some stand-alone code that does this and am happy to make the changes to `simulate_sparse_stc`.

agramfort, 2016-06-14:
fine with me. That could be a param:

location='random' | 'center'

can you open a PR?

dgwakeman, 2016-06-14:
If you are looking into this, I would personally recommend looking at doing some sort of orientation alignment and then a median based on that. Because for many labels the center of mass will have a strange orientation e.g. center of mass for the central sulcus. With that you would end up getting a radial dipole, but the more logical choice would be to do some sort of z-alignment and get a dipole oriented A-P/P-A in the middle of the humonculous. hth

Eric89GXL, 2016-06-15:
>  I would personally recommend looking at doing some sort of orientation alignment and then a median based on that.

Incorporating the orientation somehow is an interesting idea, and certainly gets you a different sort of information that may or may not be more useful depending on the use case. So I don't think it's necessary for the functionality mentioned here. I think a first pass of this "orientation-weighted center of mass" idea could be implemented fairly easily given our existing label sign flipping code, but let's discuss that in a separate issue if people are interested. I don't think it's trivial to get right unfortunately -- for example, in the given central sulcus example, I don't know which side of the sulcus the "orientation-weighted center of mass" should land on.

So for now the easiest thing would be to move a bunch of code from [SourceEstimate.center_of_mass] (https://github.com/mne-tools/mne-python/blob/master/mne/source_estimate.py#L1196) to a private function, and let `Label.center_of_mass` and `SourceEstimate.center_of_mass` both use it.

PR #3318 (2016-06-22): MRG: Add "center" option to simulate_sparse_stc
~~Needs #3314 merged first.~~

Closes #3307.
closed 2016-06-22 by agramfort