mne-tools/mne-python #3308: Tutorial on generating synthetic data with point spread

nfoti, 2016-06-14:
I have been writing some code to evaluate the effect of point-spread on inferring connectivity. @Eric89GXL suggested that a tutorial on how to put synthetic signals into a source space and corrupt those signals with point-spread via a forward and inverse solution could be useful.

I created this issue to get some feedback on what a good minimal example would be. I need to clean up my current code a bit, but can include it in the issue soon.

kingjr, 2016-06-14:
Great! I would personally expect to make two scenarios: one with independent sources and one with sync sources then:
* simulate two sources in `sample` source space
* add `1/f`source noise
* project them in sensor space with the fwd model
* add `1/f` sensor noise
* inverse the data
* apply connectivity analysis of interest

nfoti, 2016-06-20:
Thanks for the feedback.

I am shooting to keep this example simpler rather than demonstrate a full pipeline. There are already some examples of how to perform connectivity analysis with `mne-python`, and I am happy to include links to them, however, I want to focus on the basics of how to put a known signal into the `mne-python` data structures and corrupt them with point-spread. Thus, I want to avoid incorporating anything superfluous to this goal so that it's easy to follow. Users can then look to the other MNE examples to make it more sophisticated.

nfoti, 2016-07-13:
Closing since #3350 has been merged.

closed 2016-07-13 by nfoti