mne-tools/mne-python #3317: Bug in show_fiff for *-trans.fif file?

christianbrodbeck, 2016-06-17:
This looks like a bug in `show_fiff`

```bash
ISRWKAVW2269D /Applications/MNE-2.7.3-3268-MacOSX-i386/share/mne/mne_analyze/fsaverage$ mne show_fiff fsaverage-trans.fif 
Traceback (most recent call last):
  File "/Users/christian/Code/mne-python/mne/commands/mne_show_fiff.py", line 27, in <module>
    run()
  File "/Users/christian/Code/mne-python/mne/commands/mne_show_fiff.py", line 22, in run
    print(mne.io.show_fiff(args[0]))
  File "/Users/christian/Code/mne-python/mne/io/open.py", line 183, in show_fiff
    out = _show_tree(fid, tree['children'][0], indent=indent, level=0,
IndexError: list index out of range
```

Eric89GXL, 2016-06-17:
Yeah that's probably a bug, `show_fiff` shouldn't error out if the file exists. Can you make a PR?

christianbrodbeck, 2016-06-17:
Yes I'll look into it

christianbrodbeck, 2016-06-18:
This occurs only for some *-trans.fiff files and not others. What seems to be the reason is that for some trans files there are no "children" (hence the error because `tree['children']` is `[]`). I don't know enough about FIFF files to know what to do, but it seems to me that the "directory" at the root level can also contain information?

This is one that works:

```py
In [1]: from mne.io.open import fiff_open

In [2]: fname = 'test-trans.fif'

In [3]: f, tree, directory = fiff_open(fname)

In [5]: tree
Out[5]: 
{'block': 0,
 'children': [{'block': array([350], dtype=int32),
   'children': [{'block': array([107], dtype=int32),
     'children': [],
     'directory': [kind: 3506 - type: 3 - size: 4 - next: 0 - pos: 236 - data: None,
      kind: 213 - type: 33 - size: 20 - next: 0 - pos: 256 - data: None,
      kind: 213 - type: 33 - size: 20 - next: 0 - pos: 292 - data: None,
      kind: 213 - type: 33 - size: 20 - next: 0 - pos: 328 - data: None],
     'id': None,
     'nchild': 0,
     'nent': 4,
     'parent_id': None}],
   'directory': [kind: 222 - type: 35 - size: 104 - next: 0 - pos: 96 - data: None],
   'id': None,
   'nchild': 1,
   'nent': 1,
   'parent_id': None}],
 'directory': [kind: 100 - type: 31 - size: 20 - next: 0 - pos: 0 - data: None,
  kind: 101 - type: 3 - size: 4 - next: 0 - pos: 36 - data: None,
  kind: 106 - type: 3 - size: 4 - next: 0 - pos: 56 - data: None,
  kind: 108 - type: 0 - size: 0 - next: -1 - pos: 404 - data: None],
 'id': {'machid': array([-1035262720,        6388], dtype=int32),
  'secs': 1316194493,
  'usecs': 320003,
  'version': 65539},
 'nchild': 1,
 'nent': 4,
 'parent_id': None}

In [6]: import mne

In [7]: mne.read_trans(fname)
Out[7]: 
<Transform  |  head->MRI (surface RAS)>
[[ 0.99930954  0.01275934  0.0348942   0.00206991]
 [ 0.00998479  0.81240475 -0.58300853  0.01130214]
 [-0.03578702  0.58295429  0.81171638 -0.02755522]
 [ 0.          0.          0.          1.        ]]
```

And this is one that does not:

```py
In [3]: fname = 'fsaverage-trans.fif'

In [4]: f, tree, directory = fiff_open(fname)

In [6]: tree
Out[6]: 
{'block': 0,
 'children': [],
 'directory': [kind: 100 - type: 31 - size: 20 - next: 0 - pos: 0 - data: None,
  kind: 101 - type: 3 - size: 4 - next: 0 - pos: 36 - data: None,
  kind: 106 - type: 3 - size: 4 - next: 0 - pos: 56 - data: None,
  kind: 222 - type: 35 - size: 104 - next: 0 - pos: 76 - data: None,
  kind: 108 - type: 0 - size: 0 - next: -1 - pos: 196 - data: None],
 'id': {'machid': array([39770, 31849], dtype=int32),
  'secs': 1205203475,
  'usecs': 0,
  'version': 65537},
 'nchild': 0,
 'nent': 5,
 'parent_id': None}

In [8]: mne.read_trans(fname)
Out[8]: 
<Transform  |  head->MRI (surface RAS)>
[[  9.99993801e-01  -3.55761126e-03  -5.70168481e-07   1.87336397e-03]
 [  3.55187245e-03   9.98389065e-01  -5.66265360e-02  -2.87957620e-02]
 [  2.02039679e-04   5.66259325e-02   9.98395383e-01  -4.12941016e-02]
 [  0.00000000e+00   0.00000000e+00   0.00000000e+00   1.00000000e+00]]

In [19]: show_fiff(fname)
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-19-e3f0753a20a4> in <module>()
----> 1 show_fiff(fname)

//anaconda/lib/python2.7/site-packages/mne/io/open.pyc in show_fiff(fname, indent, read_limit, max_str, output, verbose)
    181     f, tree, directory = fiff_open(fname)
    182     with f as fid:
--> 183         out = _show_tree(fid, tree['children'][0], indent=indent, level=0,
    184                          read_limit=read_limit, max_str=max_str)
    185     if output == str:

IndexError: list index out of range
```

christianbrodbeck, 2016-06-18:
```py
In [10]: FIFF.FIFF_COORD_TRANS
Out[10]: 222
```
and 222 is in the root directory of the second, erroring file....

If I modify show_fiff to start with the `tree` instead of `tree['children'][0]` I get 

```bash
$ mne show_fiff test-trans.fif 
0 = ???
    100 = FIFF_FILE_ID (20) = {'secs': 1316194493, 'version' ... dict len=4
    101 = FIFF_DIR_POINTER (4) = [-1]
    106 = FIFF_FREE_LIST (4) = [-1]
    108 = FIFF_NOP (0)
    [350] = FIFFB_MNE
        222 = FIFF_COORD_TRANS (104) = <Transform  |  head->MRI (surf ... dict len=3
        [107] = FIFFB_ISOTRAK
            3506 = FIFF_MNE_COORD_FRAME (4) = [5]
            x3: 213 = FIFF_DIG_POINT (20) = {'coord_frame': 0, 'ident': 3, ... dict len=4
```
and 
```bash 
$ mne show_fiff fsaverage-trans.fif 
0 = ???
    100 = FIFF_FILE_ID (20) = {'secs': 1205203475, 'version' ... dict len=4
    101 = FIFF_DIR_POINTER (4) = [-1]
    106 = FIFF_FREE_LIST (4) = [-1]
    222 = FIFF_COORD_TRANS (104) = <Transform  |  head->MRI (surf ... dict len=3
    108 = FIFF_NOP (0)
```

Eric89GXL, 2016-06-18:
Thanks for the investigation, I think I found a reasonable fix

PR #3321 (2016-06-18): FIX: Fix show_fiff
Closes #3317.

@christianbrodbeck can you see if it fixes the issue for you?
closed 2016-06-18 by agramfort