mne-tools/mne-python #3326: ENH: New modes for extract_label_time_course

nfoti, 2016-06-21:
I'm proposing to added some new modes to the `SourceEstimate.extract_label_time_course` method and appreciate any feedback. Specifically, I would like to add `weighted-mean` and `weighted-mean-flip` modes that would compute a (flipped) weighted-mean for each label with weights given by the values of the label. This functionality would be useful for instance to weight the signals in each label proportional to their distance from the center of the label.

Eric89GXL, 2016-06-22:
Sounds reasonable to me

agramfort, 2016-06-22:
I see conceptually the reason but do you have some evidence that it's
actually useful / helpful on data?

Eric89GXL, 2016-06-22:
@wronk has some evidence coming out any day now. But even without a
publication it seems useful for aiding simulation studies, no?

agramfort, 2016-06-22:
well I am bit reluctant to add methods / options for our users unless it's really what we recommend to use. Let's try to limit confusion.

my 2c

so I did not say no but I'd like to be able to explain why we offer this option.

nfoti, 2016-06-23:
I'm find tabling this issue for now. It just seemed like an easy way to add in a lot of flexibility that has come up recently for generating synthetic data.

Eric89GXL, 2016-06-23:
Fine by me

closed 2016-06-23 by Eric89GXL